solvespace/solvespace #93: not initialized

Evil-Spirit, 2016-10-22:


Evil-Spirit, 2016-10-22:
This is also can be included https://github.com/Evil-Spirit/solvespace/commit/e07404afb8f258c7935cc53cc875a6efe37fd391

whitequark, 2016-11-17:
The current master and opengl2 branches use the style like:

    OpenGl1Renderer() : camera(), lighting(), current() {}

I cannot reproduce the bug on master or opengl2, when built with MSVC locally, or on Appveyor. Can you confirm that this problem is not present? Or show where the uninitialized data ends?

Evil-Spirit, 2016-11-17:
@whitequark,
Yes, I still have this problem on master (GL Error 0x502).
This is because current.drawing not initialized with false

whitequark, 2016-11-17:
OK, I'll look further into it.

whitequark, 2016-11-29:
@Evil-Spirit May be similar to bc62478c0bc.

Evil-Spirit, 2016-12-01:
@whitequark, yes this bugs of the same nature.

Evil-Spirit, 2017-03-02:
@whitequark also, commints from whole branch Evil-Spirit/solvespace-master/commits/93-not-initialized

whitequark, 2019-05-23:
@Evil-Spirit I finally figured it out, should be fixed in 43a59e21.

closed 2019-05-23 by whitequark