solvespace/solvespace #95: entity handles - design question

jmkinzer, 2016-10-25:
I've been reviewing the code a bit and the use of entity handles (e.g. as appears in hRequest::entity). If I understand well then (for example) the following code:
```
hRequest someRequest;
...
SK.GetEntity(someRequest.entity(1))
```
Could be restated as:
```
...
SK.GetEntity(SK.GetEntity(someRequest.entity(0))->point[0]);
```

Where the latter is technically less brittle as it doesn't assume where points are stored in the entity indices. (Of course it's slower and unwieldy...)

But if there were hypothetical helpers of the form
```
Entity& hRequest::point(int N) const
Entity& hRequest::normal(int N) const
...
```
The above becomes simply:
`someRequest.point(0)
`

Is my basic understanding correct here? Is there a reason using entity offsets as in the first form is preferred?

Thanks

jwesthues, 2016-10-26:
It's ugly that we don't use such helpers. It's not brittle, since that mapping can't ever change without breaking all nontrivial saved files.

jmkinzer, 2016-10-26:
OK - sounds like I understand things properly, which was my main concern. Thanks!

closed 2016-10-26 by jmkinzer