solvespace/solvespace #96: Mac OS crash (EXC_BAD_ACCESS)

michthom, 2016-11-05:
I'm hoping this backtrace will help find this issue, but please show me where else to look if required?

`Process 7049 stopped
* thread #1: tid = 0x431d0e, 0x0000002000000020, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x2000000020)
    frame #0: 0x0000002000000020
error: memory read failed for 0x2000000000
(lldb) thread backtrace all
* thread #1: tid = 0x431d0e, 0x0000002000000020, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x2000000020)
  * frame #0: 0x0000002000000020
    frame #1: 0x000000010003e239 solvespace`SolveSpace::TextWindow::MouseEvent(this=0x0000000102000000, leftClick=false, leftDown=false, x=-516.8359375, y=13.72265625) + 761 at textwin.cpp:1044
    frame #2: 0x000000010004b865 solvespace`::-[TextWindowView mouseMoved:](self=0x000000010102ab90, _cmd="mouseMoved:", event=0x00000001015de920) + 293 at cocoamain.mm:921
    frame #3: 0x00007fff99b631e4 AppKit`-[NSTrackingArea mouseMoved:] + 152
    frame #4: 0x00007fff9a0e9bd1 AppKit`-[NSWindow _handleMouseMovedEvent:] + 266
    frame #5: 0x00007fff9a0eab15 AppKit`-[NSWindow _reallySendEvent:isDelayedEvent:] + 2108
    frame #6: 0x00007fff99b29539 AppKit`-[NSWindow sendEvent:] + 517
    frame #7: 0x00007fff99aa94db AppKit`-[NSApplication sendEvent:] + 1167
    frame #8: 0x00007fff99910df2 AppKit`-[NSApplication run] + 796
    frame #9: 0x000000010004ee6d solvespace`main(argc=1, argv=0x00007fff5fbffb08) + 429 at cocoamain.mm:1352
    frame #10: 0x00007fff914035ad libdyld.dylib`start + 1
    frame #11: 0x00007fff914035ad libdyld.dylib`start + 1

  thread #3: tid = 0x431d2a, 0x00007fff8d933efa libsystem_kernel.dylib`kevent_qos + 10, queue = 'com.apple.libdispatch-manager'
    frame #0: 0x00007fff8d933efa libsystem_kernel.dylib`kevent_qos + 10
    frame #1: 0x00007fff8ee6f165 libdispatch.dylib`_dispatch_mgr_invoke + 216
    frame #2: 0x00007fff8ee6edcd libdispatch.dylib`_dispatch_mgr_thread + 52

  thread #12: tid = 0x431d61, 0x00007fff8d92cf72 libsystem_kernel.dylib`mach_msg_trap + 10, name = 'com.apple.NSEventThread'
    frame #0: 0x00007fff8d92cf72 libsystem_kernel.dylib`mach_msg_trap + 10
    frame #1: 0x00007fff8d92c3b3 libsystem_kernel.dylib`mach_msg + 55
    frame #2: 0x00007fff884f2124 CoreFoundation`__CFRunLoopServiceMachPort + 212
    frame #3: 0x00007fff884f15ec CoreFoundation`__CFRunLoopRun + 1356
    frame #4: 0x00007fff884f0e38 CoreFoundation`CFRunLoopRunSpecific + 296
    frame #5: 0x00007fff99a72d95 AppKit`_NSEventThread + 149
    frame #6: 0x00007fff9b60c99d libsystem_pthread.dylib`_pthread_body + 131
    frame #7: 0x00007fff9b60c91a libsystem_pthread.dylib`_pthread_start + 168
    frame #8: 0x00007fff9b60a351 libsystem_pthread.dylib`thread_start + 13

  thread #46: tid = 0x433ec6, 0x00007fff8d9335e2 libsystem_kernel.dylib`__workq_kernreturn + 10
    frame #0: 0x00007fff8d9335e2 libsystem_kernel.dylib`__workq_kernreturn + 10
    frame #1: 0x00007fff9b60c578 libsystem_pthread.dylib`_pthread_wqthread + 1283
    frame #2: 0x00007fff9b60a341 libsystem_pthread.dylib`start_wqthread + 13

  thread #51: tid = 0x4341ce, 0x00007fff8d9335e2 libsystem_kernel.dylib`__workq_kernreturn + 10
    frame #0: 0x00007fff8d9335e2 libsystem_kernel.dylib`__workq_kernreturn + 10
    frame #1: 0x00007fff9b60c578 libsystem_pthread.dylib`_pthread_wqthread + 1283
    frame #2: 0x00007fff9b60a341 libsystem_pthread.dylib`start_wqthread + 13

  thread #52: tid = 0x43425e, 0x00007fff8d9335e2 libsystem_kernel.dylib`__workq_kernreturn + 10
    frame #0: 0x00007fff8d9335e2 libsystem_kernel.dylib`__workq_kernreturn + 10
    frame #1: 0x00007fff9b60c578 libsystem_pthread.dylib`_pthread_wqthread + 1283
    frame #2: 0x00007fff9b60a341 libsystem_pthread.dylib`start_wqthread + 13

  thread #53: tid = 0x43425f, 0x00007fff8d9335e2 libsystem_kernel.dylib`__workq_kernreturn + 10
    frame #0: 0x00007fff8d9335e2 libsystem_kernel.dylib`__workq_kernreturn + 10
    frame #1: 0x00007fff9b60c578 libsystem_pthread.dylib`_pthread_wqthread + 1283
    frame #2: 0x00007fff9b60a341 libsystem_pthread.dylib`start_wqthread + 13

  thread #54: tid = 0x4342ac, 0x00007fff8d9335e2 libsystem_kernel.dylib`__workq_kernreturn + 10
    frame #0: 0x00007fff8d9335e2 libsystem_kernel.dylib`__workq_kernreturn + 10
    frame #1: 0x00007fff9b60c578 libsystem_pthread.dylib`_pthread_wqthread + 1283
    frame #2: 0x00007fff9b60a341 libsystem_pthread.dylib`start_wqthread + 13

  thread #55: tid = 0x4342ad, 0x00007fff8d9335e2 libsystem_kernel.dylib`__workq_kernreturn + 10
    frame #0: 0x00007fff8d9335e2 libsystem_kernel.dylib`__workq_kernreturn + 10
    frame #1: 0x00007fff9b60c578 libsystem_pthread.dylib`_pthread_wqthread + 1283
    frame #2: 0x00007fff9b60a341 libsystem_pthread.dylib`start_wqthread + 13

  thread #56: tid = 0x4342ae, 0x00007fff8d9335e2 libsystem_kernel.dylib`__workq_kernreturn + 10
    frame #0: 0x00007fff8d9335e2 libsystem_kernel.dylib`__workq_kernreturn + 10
    frame #1: 0x00007fff9b60c578 libsystem_pthread.dylib`_pthread_wqthread + 1283
    frame #2: 0x00007fff9b60a341 libsystem_pthread.dylib`start_wqthread + 13

  thread #57: tid = 0x4342af, 0x00007fff8d9335e2 libsystem_kernel.dylib`__workq_kernreturn + 10
    frame #0: 0x00007fff8d9335e2 libsystem_kernel.dylib`__workq_kernreturn + 10
    frame #1: 0x00007fff9b60c578 libsystem_pthread.dylib`_pthread_wqthread + 1283
    frame #2: 0x00007fff9b60a341 libsystem_pthread.dylib`start_wqthread + 13

  thread #58: tid = 0x4342b0, 0x00007fff8d9335e2 libsystem_kernel.dylib`__workq_kernreturn + 10
    frame #0: 0x00007fff8d9335e2 libsystem_kernel.dylib`__workq_kernreturn + 10
    frame #1: 0x00007fff9b60c578 libsystem_pthread.dylib`_pthread_wqthread + 1283
    frame #2: 0x00007fff9b60a341 libsystem_pthread.dylib`start_wqthread + 13
`

michthom, 2016-11-05:
Version 3.0 (3.0~ea52fcbc) compiled from source with debug symbols.

whitequark, 2016-11-05:
Please run SolveSpace under debugger and tell me the values of locals `r` and `c` at the moment of crash.

michthom, 2016-11-05:
This is the epitome of speedy service! I was running it under LLDB, so when it stopped I dumped the "thread backtrace all" above. Stupidly I didn't then write a core file to analyse. I'll try to reproduce it now.

michthom, 2016-11-05:
Reproduced below, and saved a core file this time - but I'm clueless on how to display the locals you wanted, I tried (to no avail):

(lldb) frame variable c
(lldb) frame variable r
(lldb) target variable c
error: can't find global variable 'c'
(lldb) target variable r
error: can't find global variable 'r'
(lldb) frame variable --no-args c
(lldb) frame variable --no-args r

`Process 7499 stopped
* thread #1: tid = 0x44b9e9, 0x0000002000000020, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x2000000020)
    frame #0: 0x0000002000000020
error: memory read failed for 0x2000000000
(lldb) thread backtrace all
* thread #1: tid = 0x44b9e9, 0x0000002000000020, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x2000000020)
  * frame #0: 0x0000002000000020
    frame #1: 0x000000010003e239 solvespace`SolveSpace::TextWindow::MouseEvent(this=0x0000000102800000, leftClick=false, leftDown=false, x=-1146.63671875, y=11.2109375) + 761 at textwin.cpp:1044
    frame #2: 0x000000010004b865 solvespace`::-[TextWindowView mouseMoved:](self=0x00000001015053d0, _cmd="mouseMoved:", event=0x00000001014a7750) + 293 at cocoamain.mm:921
    frame #3: 0x00007fff99b631e4 AppKit`-[NSTrackingArea mouseMoved:] + 152
    frame #4: 0x00007fff9a0e9bd1 AppKit`-[NSWindow _handleMouseMovedEvent:] + 266
    frame #5: 0x00007fff9a0eab15 AppKit`-[NSWindow _reallySendEvent:isDelayedEvent:] + 2108
    frame #6: 0x00007fff99b29539 AppKit`-[NSWindow sendEvent:] + 517
    frame #7: 0x00007fff99aa94db AppKit`-[NSApplication sendEvent:] + 1167
    frame #8: 0x00007fff99910df2 AppKit`-[NSApplication run] + 796
    frame #9: 0x000000010004ee6d solvespace`main(argc=1, argv=0x00007fff5fbffb08) + 429 at cocoamain.mm:1352
    frame #10: 0x00007fff914035ad libdyld.dylib`start + 1
    frame #11: 0x00007fff914035ad libdyld.dylib`start + 1

  thread #3: tid = 0x44ba04, 0x00007fff8d933efa libsystem_kernel.dylib`kevent_qos + 10, queue = 'com.apple.libdispatch-manager'
    frame #0: 0x00007fff8d933efa libsystem_kernel.dylib`kevent_qos + 10
    frame #1: 0x00007fff8ee6f165 libdispatch.dylib`_dispatch_mgr_invoke + 216
    frame #2: 0x00007fff8ee6edcd libdispatch.dylib`_dispatch_mgr_thread + 52

  thread #12: tid = 0x44ba24, 0x00007fff8d92cf72 libsystem_kernel.dylib`mach_msg_trap + 10, name = 'com.apple.NSEventThread'
    frame #0: 0x00007fff8d92cf72 libsystem_kernel.dylib`mach_msg_trap + 10
    frame #1: 0x00007fff8d92c3b3 libsystem_kernel.dylib`mach_msg + 55
    frame #2: 0x00007fff884f2124 CoreFoundation`__CFRunLoopServiceMachPort + 212
    frame #3: 0x00007fff884f15ec CoreFoundation`__CFRunLoopRun + 1356
    frame #4: 0x00007fff884f0e38 CoreFoundation`CFRunLoopRunSpecific + 296
    frame #5: 0x00007fff99a72d95 AppKit`_NSEventThread + 149
    frame #6: 0x00007fff9b60c99d libsystem_pthread.dylib`_pthread_body + 131
    frame #7: 0x00007fff9b60c91a libsystem_pthread.dylib`_pthread_start + 168
    frame #8: 0x00007fff9b60a351 libsystem_pthread.dylib`thread_start + 13

  thread #33: tid = 0x44d232, 0x00007fff8d9335e2 libsystem_kernel.dylib`__workq_kernreturn + 10
    frame #0: 0x00007fff8d9335e2 libsystem_kernel.dylib`__workq_kernreturn + 10
    frame #1: 0x00007fff9b60c578 libsystem_pthread.dylib`_pthread_wqthread + 1283
    frame #2: 0x00007fff9b60a341 libsystem_pthread.dylib`start_wqthread + 13
`

whitequark, 2016-11-05:
You need to go up one frame before asking lldb for this variable.

michthom, 2016-11-05:
Got it, sorry for being slow:

`$ lldb ~/Desktop/solvespace/build/src/solvespace.app/Contents/MacOS/solvespace -c /cores/2016-solvespace.core 
(lldb) target create "/Users/michthom/Desktop/solvespace/build/src/solvespace.app/Contents/MacOS/solvespace" --core "/cores/2016-solvespace.core"
Core file '/cores/2016-solvespace.core' (x86_64) was loaded.
(lldb) bt all
warning: could not load any Objective-C class information. This will significantly reduce the quality of type information available.
* thread #1: tid = 0x0000, 0x0000002000000020, stop reason = signal SIGSTOP
  * frame #0: 0x0000002000000020
    frame #1: 0x000000010003e239 solvespace`SolveSpace::TextWindow::MouseEvent(this=0x0000000102800000, leftClick=false, leftDown=false, x=-1146.63671875, y=11.2109375) + 761 at textwin.cpp:1044
    frame #2: 0x000000010004b865 solvespace`::-[TextWindowView mouseMoved:](self=0x00000001015053d0, _cmd="mouseMoved:", event=0x00000001014a7750) + 293 at cocoamain.mm:921
    frame #3: 0x00007fff99b631e4 AppKit`-[NSTrackingArea mouseMoved:] + 152
    frame #4: 0x00007fff9a0e9bd1 AppKit`-[NSWindow _handleMouseMovedEvent:] + 266
    frame #5: 0x00007fff9a0eab15 AppKit`-[NSWindow _reallySendEvent:isDelayedEvent:] + 2108
    frame #6: 0x00007fff99b29539 AppKit`-[NSWindow sendEvent:] + 517
    frame #7: 0x00007fff99aa94db AppKit`-[NSApplication sendEvent:] + 1167
    frame #8: 0x00007fff99910df2 AppKit`-[NSApplication run] + 796
    frame #9: 0x000000010004ee6d solvespace`main(argc=1, argv=0x00007fff5fbffb08) + 429 at cocoamain.mm:1352
    frame #10: 0x00007fff914035ad libdyld.dylib`start + 1
    frame #11: 0x00007fff914035ad libdyld.dylib`start + 1

  thread #2: tid = 0x0001, 0x00007fff8d933efa libsystem_kernel.dylib`kevent_qos + 10, stop reason = signal SIGSTOP
    frame #0: 0x00007fff8d933efa libsystem_kernel.dylib`kevent_qos + 10
    frame #1: 0x00007fff8ee6f165 libdispatch.dylib`_dispatch_mgr_invoke + 216
    frame #2: 0x00007fff8ee6edcd libdispatch.dylib`_dispatch_mgr_thread + 52

  thread #3: tid = 0x0002, 0x00007fff8d92cf72 libsystem_kernel.dylib`mach_msg_trap + 10, stop reason = signal SIGSTOP
    frame #0: 0x00007fff8d92cf72 libsystem_kernel.dylib`mach_msg_trap + 10
    frame #1: 0x00007fff8d92c3b3 libsystem_kernel.dylib`mach_msg + 55
    frame #2: 0x00007fff884f2124 CoreFoundation`__CFRunLoopServiceMachPort + 212
    frame #3: 0x00007fff884f15ec CoreFoundation`__CFRunLoopRun + 1356
    frame #4: 0x00007fff884f0e38 CoreFoundation`CFRunLoopRunSpecific + 296
    frame #5: 0x00007fff99a72d95 AppKit`_NSEventThread + 149
    frame #6: 0x00007fff9b60c99d libsystem_pthread.dylib`_pthread_body + 131
    frame #7: 0x00007fff9b60c91a libsystem_pthread.dylib`_pthread_start + 168
    frame #8: 0x00007fff9b60a351 libsystem_pthread.dylib`thread_start + 13

  thread #4: tid = 0x0003, 0x00007fff8d9335e2 libsystem_kernel.dylib`__workq_kernreturn + 10, stop reason = signal SIGSTOP
    frame #0: 0x00007fff8d9335e2 libsystem_kernel.dylib`__workq_kernreturn + 10
    frame #1: 0x00007fff9b60c578 libsystem_pthread.dylib`_pthread_wqthread + 1283
    frame #2: 0x00007fff9b60a341 libsystem_pthread.dylib`start_wqthread + 13
(lldb) frame sleect 1
invalid command 'frame sleect'.
(lldb) frame select 1
frame #1: 0x000000010003e239 solvespace`SolveSpace::TextWindow::MouseEvent(this=0x0000000102800000, leftClick=false, leftDown=false, x=-1146.63671875, y=11.2109375) + 761 at textwin.cpp:1044
   1041	            if(item.link) {
   1042	                SetMousePointerToHand(true);
   1043	                if(item.h) {
-> 1044	                    (item.h)(item.link, item.data);
   1045	                }
   1046	            } else {
   1047	                SetMousePointerToHand(false);
(lldb) frame variable c
(int) c = -128
(lldb) frame variable r
(int) r = 0
`

michthom, 2016-11-05:
Got it ( at last! )

$ lldb ~/Desktop/solvespace/build/src/solvespace.app/Contents/MacOS/solvespace -c /cores/2016-solvespace.core 
(lldb) target create "/Users/michthom/Desktop/solvespace/build/src/solvespace.app/Contents/MacOS/solvespace" --core "/cores/2016-solvespace.core"
Core file '/cores/2016-solvespace.core' (x86_64) was loaded.
(lldb) bt all
warning: could not load any Objective-C class information. This will significantly reduce the quality of type information available.
* thread #1: tid = 0x0000, 0x0000002000000020, stop reason = signal SIGSTOP
  * frame #0: 0x0000002000000020
    frame #1: 0x000000010003e239 solvespace`SolveSpace::TextWindow::MouseEvent(this=0x0000000102800000, leftClick=false, leftDown=false, x=-1146.63671875, y=11.2109375) + 761 at textwin.cpp:1044
    frame #2: 0x000000010004b865 solvespace`::-[TextWindowView mouseMoved:](self=0x00000001015053d0, _cmd="mouseMoved:", event=0x00000001014a7750) + 293 at cocoamain.mm:921
    frame #3: 0x00007fff99b631e4 AppKit`-[NSTrackingArea mouseMoved:] + 152
    frame #4: 0x00007fff9a0e9bd1 AppKit`-[NSWindow _handleMouseMovedEvent:] + 266
    frame #5: 0x00007fff9a0eab15 AppKit`-[NSWindow _reallySendEvent:isDelayedEvent:] + 2108
    frame #6: 0x00007fff99b29539 AppKit`-[NSWindow sendEvent:] + 517
    frame #7: 0x00007fff99aa94db AppKit`-[NSApplication sendEvent:] + 1167
    frame #8: 0x00007fff99910df2 AppKit`-[NSApplication run] + 796
    frame #9: 0x000000010004ee6d solvespace`main(argc=1, argv=0x00007fff5fbffb08) + 429 at cocoamain.mm:1352
    frame #10: 0x00007fff914035ad libdyld.dylib`start + 1
    frame #11: 0x00007fff914035ad libdyld.dylib`start + 1

  thread #2: tid = 0x0001, 0x00007fff8d933efa libsystem_kernel.dylib`kevent_qos + 10, stop reason = signal SIGSTOP
    frame #0: 0x00007fff8d933efa libsystem_kernel.dylib`kevent_qos + 10
    frame #1: 0x00007fff8ee6f165 libdispatch.dylib`_dispatch_mgr_invoke + 216
    frame #2: 0x00007fff8ee6edcd libdispatch.dylib`_dispatch_mgr_thread + 52

  thread #3: tid = 0x0002, 0x00007fff8d92cf72 libsystem_kernel.dylib`mach_msg_trap + 10, stop reason = signal SIGSTOP
    frame #0: 0x00007fff8d92cf72 libsystem_kernel.dylib`mach_msg_trap + 10
    frame #1: 0x00007fff8d92c3b3 libsystem_kernel.dylib`mach_msg + 55
    frame #2: 0x00007fff884f2124 CoreFoundation`__CFRunLoopServiceMachPort + 212
    frame #3: 0x00007fff884f15ec CoreFoundation`__CFRunLoopRun + 1356
    frame #4: 0x00007fff884f0e38 CoreFoundation`CFRunLoopRunSpecific + 296
    frame #5: 0x00007fff99a72d95 AppKit`_NSEventThread + 149
    frame #6: 0x00007fff9b60c99d libsystem_pthread.dylib`_pthread_body + 131
    frame #7: 0x00007fff9b60c91a libsystem_pthread.dylib`_pthread_start + 168
    frame #8: 0x00007fff9b60a351 libsystem_pthread.dylib`thread_start + 13

  thread #4: tid = 0x0003, 0x00007fff8d9335e2 libsystem_kernel.dylib`__workq_kernreturn + 10, stop reason = signal SIGSTOP
    frame #0: 0x00007fff8d9335e2 libsystem_kernel.dylib`__workq_kernreturn + 10
    frame #1: 0x00007fff9b60c578 libsystem_pthread.dylib`_pthread_wqthread + 1283
    frame #2: 0x00007fff9b60a341 libsystem_pthread.dylib`start_wqthread + 13
(lldb) frame select 1
invalid command 'frame select'.
(lldb) frame select 1
frame #1: 0x000000010003e239 solvespace`SolveSpace::TextWindow::MouseEvent(this=0x0000000102800000, leftClick=false, leftDown=false, x=-1146.63671875, y=11.2109375) + 761 at textwin.cpp:1044

```
   1041	            if(item.link) {
   1042	                SetMousePointerToHand(true);
   1043	                if(item.h) {
-> 1044	                    (item.h)(item.link, item.data);
   1045	                }
   1046	            } else {
   1047	                SetMousePointerToHand(false);

```

(lldb) frame variable c
(int) c = -128
(lldb) frame variable r
(int) r = 0
(lldb) frame variable item.h
(LinkFunction *) item.h = 0x0000002000000020
(lldb) frame variable item.link
(int) item.link = 32
(lldb) frame variable item.data
(uint32_t) item.data = 32

whitequark, 2016-11-05:
Ah yes... `c = -128`.

whitequark, 2016-11-05:
Fixed in master and 2.x.

closed 2016-11-05 by whitequark

michthom, 2016-11-05:
I have no idea what your c=-128 comment actually meant, but as I read it I heard the line from Kung Fu Panda - "Ah, my old enemy... stairs." :-)