solvespace/solvespace #102: OpenGL 2 renderer

whitequark, 2016-11-16:
This is the tracking issue for the OpenGL 2 renderer, being prepared in the `opengl2` branch.

There are currently the following problems with the renderer:
  - [ ] face selection is broken
  - [ ] display of triangle mesh of solid model is broken
  - [ ] normals are inverted (?) on Linux/Intel/Mesa graphics stack
  - [ ] SolveSpace crashes at startup when used with Optimus on Linux

Evil-Spirit, 2016-11-17:
@whitequark,
1)https://github.com/Evil-Spirit/solvespace/commit/3929a18c7e65c92ccf6d855bcc1027b8aa235457
2)https://github.com/Evil-Spirit/solvespace/commit/50c773af163ead147f86acf2d8e642932c208b82

whitequark, 2016-11-18:
@Evil-Spirit I figured out why I had to do glFrontFace. This is because when we do `UpdateProjection(/*flip=*/true)`, our projection matrix inverts the handedness of the coordinate system. And this worked in OpenGL 1 because it does culling with modelview matrix, and in our case it does culling with modelview*projection matrix.

whitequark, 2016-11-18:
@Evil-Spirit I've updated the opengl2 branch. There's one last bug, back faces aren't actually drawn for some reason...

Evil-Spirit, 2016-11-18:
@whitequark,
Great. I will look into back faces.

whitequark, 2016-11-18:
@Evil-Spirit I've already fixed that actually

Evil-Spirit, 2016-11-18:
@whitequark, I mean "back faces aren't drawn, even when enabled"

whitequark, 2016-11-18:
@Evil-Spirit Ah yeah, please look into it

whitequark, 2016-11-18:
@Evil-Spirit Another problem: on OS X with 3.2 core profile enabled, GL returns 0x502 at the following line in MeshRenderer::Draw:
```
glEnableVertexAttribArray(ATTRIB_POS);
```
(and I think all other glEnableVertexAttribArray calls). Do you forget to create the names or something?

whitequark, 2016-11-18:
@Evil-Spirit OK I figured it out, GL 3.2+ Core Profile requires VAOs to be used (https://stackoverflow.com/questions/24643027/opengl-invalid-operation-following-glenablevertexattribarray,  https://stackoverflow.com/questions/5970087/understanding-vertex-array-objects-glgenvertexarrays). On the other hand, OS X in Compatibility Profile allows *only* `#version 120` shaders to be used.

Wow, what a trash fire...

Evil-Spirit, 2016-11-18:
@whitequark,
We are targeting GLES 2.0 - compatible or I miss something?

whitequark, 2016-11-18:
@Evil-Spirit We are, but OS X cannot render GLES 2.0, so workarounds are needed. But I have thought up  how to do it in only a mildly horrible way.

Evil-Spirit, 2016-11-18:
What about Angle for OsX?

whitequark, 2016-11-18:
It's possible but I would like to avoid doing all this more complicated than it already is...

whitequark, 2016-11-18:
@Evil-Spirit I had to do this hack: https://github.com/solvespace/solvespace/blob/opengl2/src/render/gl2shader.cpp#L63-L87

Evil-Spirit, 2016-11-18:
@whitequark,
I think this is no so bad solution.

whitequark, 2016-11-18:
@Evil-Spirit I fixed the back face problem. It was because SSurface::TriangulateInto first filled the mesh with triangles with empty color, and then filled the color in separately.

whitequark, 2016-11-18:
this is finally **MERGED**

closed 2016-11-18 by whitequark