solvespace/solvespace #103: Unability to change the origin of a workplane

lineprinter, 2016-11-17:


wpwrak, 2017-02-09:
The only reason I experienced so far for wanting to change the origin of a workplane was that I had picked a point at the wrong level for linear extruding. If two-sided extrusions would allow the user to place both end faces, not just one with the other inevitably being symmetrical to the workplane, extrusions would no longer have to begin at a workplane. Thus, the origin would be largely irrelevant.

Maybe it might be less time-consuming to add asymmetric extrusion than to implement reassignment of the origin ?

aep, 2019-08-24:
what do you use as a workaround? copy paste all points to a new group?

gituuu, 2020-03-16:
Just commenting that asymmetrical extrusion would be really useful. It would save copy and pasting to a new group like you suggest.

ruevs, 2020-10-20:
There is another nice hint how to construct moveable work planes here:
https://github.com/solvespace/solvespace/issues/738#issuecomment-713138016

polyfloyd, 2024-04-20:
+1 from me for this feature. I encountered this last week when I had to move a group relative to another group, which caused a very crucial workplane to be at the wrong level.

For others looking for a remedy in the meantime, I resorted to hacking the file with a text editor.