solvespace/solvespace #112: Internationalization

whitequark, 2016-11-17:
This requires:
  * writing a gettext runtime implementation that's not as gross as gettext itself (and works properly on win32 anyhow, and doesn't need po→mo translation);
  * marking menus and the like with `_(...)` (easy);
  * localizing the Printf-based text window (hard).

Symbian9, 2016-11-27:
Maybe simple textfiles, similar to how its realised in PolygonCrest and YSFlight?
* https://github.com/YSFlight-opensource/PolygonCrest
* https://github.com/YSFlight-opensource/ysflight-languages

whitequark, 2016-11-27:
No, the gettext format will be used. The advantage is that there is a lot of existing tools, like xgettext and poedit, that understand them.

whitequark, 2017-01-07:
@Symbian9 I've applied your ua-UA translation. I cannot reply to your email because your mail host rejects my mail with:
```
<appsoft@ua.fm>: host mx6.ua.fm[91.198.36.62] said: 550 Host in a black list.
    See http://mail.i.ua/err/9/?ip=188.166.218.19 (in reply to RCPT TO command)
Reporting-MTA: dns; uruz.whitequark.org
X-Postfix-Queue-ID: 2EFC9203A9
X-Postfix-Sender: rfc822; whitequark@whitequark.org
Arrival-Date: Fri,  6 Jan 2017 22:29:51 -0500 (EST)

Final-Recipient: rfc822; appsoft@ua.fm
Action: failed
Status: 5.0.0
Remote-MTA: dns; mx6.ua.fm
Diagnostic-Code: smtp; 550 Host in a black list. See
    http://mail.i.ua/err/9/?ip=188.166.218.19
```

Symbian9, 2017-01-07:
@whitequark, thanks! 

Think, problem is with your own mail server, that not recognised by my mail hosting.
Try use <code>Symbian9@github.com</code>
* https://developer.github.com/v3/users/emails/

whitequark, 2017-01-07:
> Think, problem is with your own mail server, that not recognised by my mail hosting.

No. Every other mail hoster so far had no problems with my server. Complain to your hoster.

> It should be uk_UA!!!

Fixed

Symbian9, 2017-01-07:
> No. Every other mail hoster so far had no problems with my server. Complain to your hoster.

@whitequark, When somebody send me message from GMail I has on problems with getting them. Also, Github & Reddit, governmental mail servers normally send mails to described above my mail host. So, its only issue in your mail hoster, that prevent sent messages from you to me.

whitequark, 2017-01-07:
> So, its only issue in your mail hoster, that prevent sent messages from you to me.

No. There is no reason for ua.fm to reject my mail, and I don't care what the excuses are.

altemirjosecoelho, 2026-10-17:
Hi! I just opened #1791 with a complete pt_BR translation of `messages.pot`.

While testing it I ran into the point this issue already lists as the hard part: the text window. `textscreens.cpp`, `confscreen.cpp`, `describescreen.cpp` and `style.cpp` pass around 330 English literals straight to `Printf`, so none of them reach the translation system in any locale (the German, Spanish and Chinese builds show the same English property browser).

Would you accept a PR that marks those strings with `_()` / `C_()` and regenerates `messages.pot`? To keep the review manageable I would start with `textscreens.cpp` only, as a proof of concept, so we can agree on how to handle the fixed-width column alignment and the `Printf` format codes (`%Fd`, `%Lv`, etc.) before touching the other three files. Happy to adapt the approach to whatever you prefer.

phkahler, 2026-10-17:
@altemirjosecoelho  the issue with the text window is that a large portion of the strings include formatting characters. We don't want those going in the translation .po files so they'll have to be broken up. Just breaking the strings would make for some very messy code. Of course there are potential issues with the layout changing based on string lengths.

Not trying to justify the app being half translated, just saying no to i18n the format strings.