solvespace/solvespace #113: 3D DXF import

whitequark, 2016-11-17:
Requested in http://solvespace.com/forum.pl?action=viewthread&parent=1679.

whitequark, 2016-11-17:
@Evil-Spirit How hard is it to implement this?

Evil-Spirit, 2016-11-17:
@whitequark, I think if libdxfrw is reading this well, we can easily support it.

whitequark, 2016-11-18:
@Evil-Spirit yes, please look into it. This seems easy enough and potentially very useful.

Evil-Spirit, 2016-11-19:
@whitequark ,
supported here
https://github.com/Evil-Spirit/solvespace/commit/f60d2e08c50c5b61dbbb43f4a20b7226e79831d8
Before importing, workplane group needs to be switched to 3d by pressing '3' key.

whitequark, 2016-11-19:
@Evil-Spirit 

> I think any entities imported from a DXF file where Z coordinate is not zero should be marked as construction.

whitequark, 2016-11-19:
Or at least the group should be switched to drawing in 3d automatically...

Evil-Spirit, 2016-11-19:
@whitequark,
Hm.. if switch to 3d there are no any reason to make in-plane things as not construction..
So what about:
Change sketch 2d->3d and
* make all as construction
* or leave all as non-construction?

Evil-Spirit, 2016-11-19:
@whitequark,
Hm... but what if you need to import some file in workplane, but this have damaged z coord? May be just show warning:"You have imported 3d data in 2d workplane. To import file as 3d you should choose Sketch->anywhere in 3d"?

whitequark, 2016-11-19:
@Evil-Spirit That sounds like it will create more problems than it will solve. The case where a 3D DXF file will read correctly with Z coordinate thrown out seems rare to me. This is more likely to squash entities on top of one another, leading to weird problems with overlapping segments, etc.

Also, even in "draw in 3d" mode, if the current group is a workplane group, it will complain about "not closed contour" or "not coplanar" etc so we still need to mark as construction.

Evil-Spirit, 2016-11-19:
@whitequark,
It seems we have to check all the points before we can decide what type of workplane we have. If we switch only when we get z!=0, we will get some firts part of entities in 2d and some in 3d (even if they are actually in 2d).

whitequark, 2016-11-19:
@Evil-Spirit Yep

Evil-Spirit, 2016-11-19:
@whitequark,
So what to do with?

whitequark, 2016-11-19:
@Evil-Spirit Check the points in a first pass, I think.

Evil-Spirit, 2016-11-19:
@whitequark, ok

Evil-Spirit, 2016-11-21:
@whitequark,
implemented here
https://github.com/Evil-Spirit/solvespace/commit/af88466963ed3719657c315b8e574d46de64588f

whitequark, 2016-11-26:
@Evil-Spirit Which 3d DXF file were you using for testing?

whitequark, 2016-11-26:
Oh, I can save one with SolveSpace.

Evil-Spirit, 2016-11-26:
@whitequark, I have used file from forum

whitequark, 2016-11-26:
@Evil-Spirit Merged, but with changes, see 41365c5f and 8f93136d.

closed 2016-11-26 by whitequark