solvespace/solvespace #114: Flickering when resizing a window

whitequark, 2016-11-18:
After the ANGLE merge, it looks like (I think) WM_SIZE/WM_PAINT are not handled properly, and when the window is resized, there is awful flickering. There is also this awful hack in w32main.cpp that I think is also related:

```
    // Repaint one more time, after we've set everything up.
    PaintGraphics();
    PaintTextWnd();
```

@Evil-Spirit please look into this.

Evil-Spirit, 2016-11-19:
@whitequark,
This problem is with ANGLE itself,
The following chages makes things better:
```
 src/libANGLE/renderer/d3d/SurfaceD3D.cpp | 3 +--
 1 file changed, 1 insertion(+), 2 deletions(-)

diff --git a/src/libANGLE/renderer/d3d/SurfaceD3D.cpp b/src/libANGLE/renderer/d3d/SurfaceD3D.cpp
index 93f6990..0b3199b 100644
--- a/src/libANGLE/renderer/d3d/SurfaceD3D.cpp
+++ b/src/libANGLE/renderer/d3d/SurfaceD3D.cpp
@@ -236,8 +236,6 @@ egl::Error SurfaceD3D::swapRect(EGLint x, EGLint y, EGLint width, EGLint height)
         }
     }
 
-    checkForOutOfDateSwapChain();
-
     return egl::Error(EGL_SUCCESS);
 }
 
@@ -279,6 +277,7 @@ bool SurfaceD3D::checkForOutOfDateSwapChain()
 
 egl::Error SurfaceD3D::swap()
 {
+    checkForOutOfDateSwapChain();
     return swapRect(0, 0, mWidth, mHeight);
 }
 
``

whitequark, 2016-11-19:
Merged, thanks

closed 2016-11-19 by whitequark