solvespace/solvespace #115: Normals are see-through

whitequark, 2016-11-18:
This is some very weird bug, probably related to depth buffer? Draw a torus (
[torus.zip](https://github.com/solvespace/solvespace/files/599891/torus.zip). You can see the normal from the circle that's used to make the torus from this angle:
[image]
... but not from a slightly different one:
[image]

@Evil-Spirit please look into this

Evil-Spirit, 2016-11-18:
@whitequark, yes, we have it everywhere. probably this is because tricks with depth buffer in ssglDepthRange

whitequark, 2016-11-18:
@Evil-Spirit I'm not sure. This is not a problem in version 2.0, 2.1 or 2.2, but it is a problem with 3.0 with OpenGL 1 and OpenGL 2. It was introduced somehow during the Canvas changes I think.

whitequark, 2016-11-18:
But 2.0 still uses the ssglDepthRange hacks.

whitequark, 2016-11-18:
@Evil-Spirit Also, this only seems to happen to normals, and not to edges...

Evil-Spirit, 2016-11-19:
fixed in https://github.com/Evil-Spirit/solvespace/commit/832a8bdf670f4f148415f8f30e359df972e7f3aa

Evil-Spirit, 2016-11-19:
@whitequark, this happen because AlignToPixelGrid uses this function.

whitequark, 2016-11-19:
Merged, thanks

closed 2016-11-19 by whitequark