solvespace/solvespace #117: TTF_TEXT broken when extruded.

Evil-Spirit, 2016-11-21:
fixed here https://github.com/Evil-Spirit/solvespace/commit/b3f8ecc113a9404f4db620e0e282998fe75ede78

whitequark, 2016-11-21:
Merged

whitequark, 2016-11-21:
@Evil-Spirit Actually, your fix doesn't work, it still crashes:
```
File /home/whitequark/Work/solvespace/src/dsc.h, line 321, function FindById:
Assertion 'Cannot find handle' failed: ((t != NULL) == false).
Backtrace:
 0: ./build/src/solvespace(_ZN10SolveSpace14assert_failureEPKcjS1_S1_S1_+0x78) [0x562ea23f8d8b]
 1: ./build/src/solvespace(_ZN10SolveSpace6IdListINS_6EntityENS_7hEntityEE8FindByIdES2_+0x5a) [0x562ea23c2dbc]
 2: ./build/src/solvespace(_ZN10SolveSpace6Sketch9GetEntityENS_7hEntityE+0x24) [0x562ea23c24f2]
 3: ./build/src/solvespace(_ZNK10SolveSpace10EntityBase24PointGetExprsInWorkplaneENS_7hEntityEPPNS_4ExprES4_+0xa0) [0x562ea24173b8]
 4: ./build/src/solvespace(_ZNK10SolveSpace10EntityBase24PointGetExprsInWorkplaneENS_7hEntityE+0x2e) [0x562ea24174ee]
 5: ./build/src/solvespace(_ZNK10SolveSpace10EntityBase17GenerateEquationsEPNS_6IdListINS_8EquationENS_9hEquationEEE+0x276) [0x562ea2419328]
 6: ./build/src/solvespace(_ZN10SolveSpace6System23WriteEquationsExceptForENS_11hConstraintEPNS_5GroupE+0x155) [0x562ea24ba3ed]
 7: ./build/src/solvespace(_ZN10SolveSpace6System5SolveEPNS_5GroupEPiPNS_4ListINS_11hConstraintEEEbb+0x4b) [0x562ea24ba5b3]
 8: ./build/src/solvespace(_ZN10SolveSpace12SolveSpaceUI10SolveGroupENS_6hGroupEb+0x1d7) [0x562ea24434a3]
 9: ./build/src/solvespace(_ZN10SolveSpace12SolveSpaceUI19SolveGroupAndReportENS_6hGroupEb+0x29) [0x562ea244321f]
10: ./build/src/solvespace(_ZN10SolveSpace12SolveSpaceUI11GenerateAllENS0_8GenerateEbb+0x648) [0x562ea244284c]
11: ./build/src/solvespace(_ZN10SolveSpace12SolveSpaceUI11GenerateAllENS0_8GenerateEbb+0x2d1) [0x562ea24424d5]
12: ./build/src/solvespace(_ZN10SolveSpace5Group9MenuGroupENS_7CommandE+0x102d) [0x562ea244c869]
13: ./build/src/solvespace(_ZN10SolveSpace12MainMenuItemIN3Gtk8MenuItemEE11on_activateEv+0x7d) [0x562ea23e1e7d]
14: /usr/lib/x86_64-linux-gnu/libgtkmm-2.4.so.1(_ZN3Gtk14MenuItem_Class17activate_callbackEP12_GtkMenuItem+0x50) [0x7face8933240]
15: /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(g_closure_invoke+0x145) [0x7face68f4f75]
16: /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(+0x2195c) [0x7face690695c]
17: /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(g_signal_emit_valist+0xe3c) [0x7face690fbcc]
18: /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(g_signal_emit+0x8f) [0x7face690ffaf]
19: /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0(+0x24801c) [0x7face804c01c]
20: /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(g_closure_invoke+0x145) [0x7face68f4f75]
21: /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(+0x21f82) [0x7face6906f82]
22: /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(g_signal_emit_valist+0x8df) [0x7face690f66f]
23: /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(g_signal_emit+0x8f) [0x7face690ffaf]
24: /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0(gtk_accel_group_activate+0xaf) [0x7face7e7387f]
25: /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0(gtk_accel_groups_activate+0xdd) [0x7face7e74cad]
26: /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0(gtk_window_activate_key+0x176) [0x7face8063e06]
27: /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0(+0x25fe71) [0x7face8063e71]
28: /usr/lib/x86_64-linux-gnu/libgtkmm-2.4.so.1(_ZN3Gtk6Widget18on_key_press_eventEP12_GdkEventKey+0x44) [0x7face89935c4]
29: ./build/src/solvespace(_ZN10SolveSpace17GraphicsWindowGtk18on_key_press_eventEP12_GdkEventKey+0x12d) [0x562ea23d6aed]
30: /usr/lib/x86_64-linux-gnu/libgtkmm-2.4.so.1(_ZN3Gtk12Widget_Class24key_press_event_callbackEP10_GtkWidgetP12_GdkEventKey+0x54) [0x7face89955e4]
31: /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0(+0x13184c) [0x7face7f3584c]
32: /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(g_closure_invoke+0x145) [0x7face68f4f75]
33: /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(+0x2237d) [0x7face690737d]
34: /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(g_signal_emit_valist+0x8df) [0x7face690f66f]
35: /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(g_signal_emit+0x8f) [0x7face690ffaf]
36: /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0(+0x24998c) [0x7face804d98c]
37: /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0(gtk_propagate_event+0x19f) [0x7face7f340ef]
38: /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0(gtk_main_do_event+0x2cb) [0x7face7f343cb]
39: /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0(+0x5acec) [0x7face795ccec]
40: /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x2a7) [0x7face29897f7]
41: /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x4aa60) [0x7face2989a60]
42: /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_loop_run+0xc2) [0x7face2989d82]
43: /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0(gtk_main+0xb7) [0x7face7f33447]
44: /usr/lib/x86_64-linux-gnu/libgtkmm-2.4.so.1(_ZN3Gtk4Main3runERNS_6WindowE+0x121) [0x7face89313c1]
45: ./build/src/solvespace(main+0x40b) [0x562ea23d45d6]
46: /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf1) [0x7face53302b1]
47: ./build/src/solvespace(_start+0x2a) [0x562ea23bbb0a]
Aborted
```

Evil-Spirit, 2016-11-21:
Hm. I have no such problem.

whitequark, 2016-11-21:
Fixed again, properly.

closed 2016-11-21 by whitequark