solvespace/solvespace #118: Allow coping of DATUM_POINT

Evil-Spirit, 2016-11-21:
There was no ability to copy/paste DATUM_POINTS
https://github.com/Evil-Spirit/solvespace/commit/983f2520991b4ad3570254534e57d4611157d8af

whitequark, 2016-11-25:
@Evil-Spirit I feel like there's inconsistency here. For example you can copy any lines (from request, imported, from extrusion...) and not just lines from requests. So it would be strange if e.g. datum points from requests could be copied but translated datum points could not. But it would also be strange if we always copied every point because then Ctrl+A, Ctrl+C, Ctrl+V would produce an explosion of points.

What about copying:
  * any selected point,
  * unless it is already from a request in the clipboard?
This seems like it will avoid both cases of strangeness.

Evil-Spirit, 2016-11-26:
@whitequark, yes, you are right, we should copy all the points except points from requests which is in copy list.

Evil-Spirit, 2016-12-19:
@whitequark, done.

Evil-Spirit, 2016-12-19:
@whitequark,
also, here is construction fix https://github.com/Evil-Spirit/solvespace-master/commit/e197934596966d53ac686b670a9e9c5024a4921e

whitequark, 2017-01-14:
Merged in master.

closed 2017-01-14 by whitequark