solvespace/solvespace #120: Regression with translate+extrude in 2.1, 2.2

ruevs, 2016-11-25:
The attached trivial sketch shows a regression in 2.1 and 2.2 releases.

The circle in g003 is translated twice in g005 and in g004.
The two translated groups are extruded in g007 and g008.
The extrude in g007 is visible only when it is selected as a working group. when any of the later groups are selected the holes disappear.

This is a regression in 2.1/2.2 if I open the file in 2.0 the holes from both extrusions are visible.

whitequark, 2016-11-25:
> Github does not allow attaching a zip file

GitHub does in fact allow attaching zip files, I do that all the time...

ruevs, 2016-11-25:
It does not allow me.
"We don’t support that file type. Try again with a PNG, GIF, JPG, DOCX, PPTX, XLSX, TXT, PDF, or ZIP. "
:-D

whitequark, 2016-11-25:
[bug.zip](https://github.com/solvespace/solvespace/files/613625/bug.zip)

whitequark, 2016-11-25:
Basically this boils down to: what happens if there are groups between source group (for some solid operation) and the solid operation group?

  * Clearly the entities should be generated from the source group.
  * Right now the source mesh/shell that undergoes the solid operation is also taken from the source group. But arguably it should be taken from the previous group in order.

I'm hesitant to just change where the mesh/shell are taken from because this might break some old files. I'll need to think about it.

ruevs, 2016-11-25:
That explains it.

What is the correct/logical behavior is hard to say. However the example came from a real detail I was modelling and I can not think of an (easy) way to do it in 2.2 with the current logic. I ended up deleting all the extrusions, opening the file in 2.0 and re-creating them - which placed them at the end.

But if you open bug.slvs in 2.0 (or 1.9 that's the oldest I have handy) the holes from both translates are present. That means that in 2.0 the logic is that the mesh undergoing the solid operation is taken from the previous group. Another interesting thing is that after complaining about unknown/newer format 2.0 rearranges the groups in order of creation. I discovered this only when creating the example file to report the bug - too late for the real model :-)

whitequark, 2016-11-25:
> That means that in 2.0 the logic is that the mesh undergoing the solid operation is taken from the previous group.

No. This means that 2.0 does not treat the files saved in 2.1 that have groups inserted in the middle like 2.1 does. This is in fact mentioned in CHANGELOG. The logic of mesh operations is exactly the same, only the group ordering has changed.

Evil-Spirit, 2016-12-18:
@whitequark,
Actually here is a more complicated problem than just group ordering.
The root of the problem is in finding RunningMeshGroup.
What actually happens when we searching the group to combine with the current one:
1. If we have simple group (not copy and transform), we just call RunningMeshGroup for the current group, which will return us :
(a) the previous group in case where the prevoius group is also simple. 
(b) If the previous group is complex (cpy-n-tf) it will return us RunningMeshGroup for complex group source (opA). So, if previous group of opA is also complex, it will iterate opA->prev->opA->prev until it found some simple group.

2. If we have complex group, we take souce group(opA) and call RunningMeshGroup for it. This is not necessary because we just can call RunningMeshGroup and this will do the same job a little bit more operations, but less confusion on code reading.

I am using simplier files:
g002 - rectangle
g003 - extrude g002
g004 - circle on g003 edge
g005 - translate g004
g006 - translate g004
g007 - extrude g005
g008 - extude g006

So, what is actually happened with this files:
for ssv20 we have:
Let's try to find running mesh for the last extrude group for v20:
This is case 1(a) (the previous extrude of the last extrude and this is right choice)
[image]

for ssv21+ we have:
This is case 1(a) for last group, we choose the previous which is translate(g005).
for complex translate we have to choose the source opA and call RunningMeshGroup. So, we choosing
g004-circle, which contains only extuded rectangle without holes copied from g003.
[image]

Evil-Spirit, 2016-12-19:
Actually, I've found some cases where the same problem appears:
https://github.com/whitequark/solvespace-testcases/tree/master/bugs/120
cube.slvs and cube2.slvs

Evil-Spirit, 2016-12-19:
@whitequark, some improvement is here

Evil-Spirit, 2016-12-19:
@whitequark,
I think there is no obvious solution here using the current approach,
We need to analyze whole group tree and model building process can't be linear step-by-step.

Evil-Spirit, 2016-12-23:
@whitequark, New thoughts.
Solvespace have linear view of groups, but actually we need to talk about tree of operations. For example we can take n-d array of boxes, then we choose some of middle group and make step-n-tf gruop. We get one root group and two child groups which can be built the same way.

The current approach with reordering groups is not breaks anything, but just shows the old problem: when we have real tree of operations, we can't build resulting model the right way, we can build only one branch, because we regenerate them one-by-one, in linear order.

Instead of this, we need to work with full-tree model building algorithm, which should work in the following way:
1. Make all the GenerateALL work on construction (entities, constraints solving, bbox etc).
2. Pass the tree from root to children and build for each group something we previously call thisMesh, but using the different approach: we create only new and unique geometry generated by the current group (like option "with copy #1" for step-n-tf groups)
3. Pass the tree from leaves to root and perform boolean operations on it. When we reach the root of two children, we perform merging the subtrees. If model contains not separated trees, we should process them in order of root groups.

whitequark, 2016-12-24:
@ruevs OK. My decision is to not classify this as regression. The rationale is as follows:
  1. The files created in 2.0 open correctly.
  2. The feature introduced in 2.1 is working exactly as planned.
  3. This is not something that can be easily fixed in 2.3--indeed it will likely require major changes--and the feature is desirable on its own, so I am not going to revert it.
  4. We need to release 2.3 as soon as possible to ensure it goes into next Debian stable.

However, I recognize that, while 2.1 adds support for drawing geometry in certain ways, it inadvertly prevents you from drawing geometry in other way (see also https://github.com/solvespace/solvespace/issues/104), and I will explore the ways to bring back the functionality you want in 3.0+.

Denominating from 2.3.

ruevs, 2016-12-25:
@Whitequark:
The reasoning makes sense.
As for the lost functionality - is there a graceful way to do it in 2.1+ ? I may be missing something?

whitequark, 2016-12-25:
@ruevs Sure, you can draw the original sketch you would like to translate twice (if it's just a circle), or you could draw it once, then import twice.