solvespace/solvespace #121: Implement tool → select → confirm workflow

whitequark, 2016-11-26:
Originally filed by @pimdegroot:

>The current way to use a tool in solvespace is to select the items you want to modify and then apply a tool.
>
>In other cad programs there is a second workflow where you select a tool, select the items you want to apply it to and then confirm.
>
>One example would be the dimension tool, which in the new workflow could work like this: Nothing selected > press D > select one or multiple lines or points > confirm with enter > have text floating under pointer > confirm location with click.
>
>Extrude could work like this: nothing selected > extrude tool > select areas to extrude > confirm > confirm location with click.

phkahler, 2022-01-18:
I'm curious how many people actually want this. It's a second way to do the same things, but actually takes *more* clicks than the existing way. We have "noun, verb" today but this asks for "verb, noun, confirm" to also be available. I honestly don't see the advantage and some improvements have been delayed/rejected in part to wait for this to land first.

But rather than just closing, I'm asking how many people would actually want this or not?

ruevs, 2022-01-20:
@phkahler No one wants is currently it seems. (neither do I)

But since there is even a (not very good) pull request on this I's leave it open for now...

Thovthe, 2022-04-15:
If you're used to verb>noun it will trip you up a bit when starting to use solvespace but this is a minor quirk.  This is low priority.

To track demand for this feature:
:thumbsup: if you want this added; :eyes: **Eyeballs** if you don't care; :thumbsdown: if it makes you feel icky gross that there be another option

closed 2023-07-26 by phkahler