solvespace/solvespace #123: SketchUp-like navigation

whitequark, 2016-11-27:
I.e. navigation where the Z axis always remains vertical during rotation. This seems helpful to novice users.

Symbian9, 2016-12-02:
I agree with Z must be vertical, but, as for me, it would be better look onto Blender interaction. Some futures and interface ideas from Blender can be usefull for SolveSpace.
* https://www.blender.org/manual/editors/3dview/navigate/index.html
* http://blender.stackexchange.com/questions/1511/how-can-i-control-the-viewport-without-a-mouse
* https://wiki.blender.org/index.php/Doc:2.4/Manual/3D_interaction/Navigating
* https://xxx.youtube.com/watch?v=zHdTUJBs7so

dynamodan, 2016-12-16:
I got here by the exact same need -- it's really a user preference and some people might want the camera standing on its head.

But I like the camera to stay locked so the "horizon" is always level.  To me it makes no sense to have the viewport gradually falling over sideways as I'm trying to navigate.

Blender calls this setting "Turntable" and it's found in File -> User Preferences -> Input -> Orbit Style.  There are two choices, Turntable and Trackball.  SolveSpace's current behaviour resembles Blender's "Trackball" orbit style.  It would be *really* nice to have Turntable in SolveSpace.

Thanks so much, @whitequark for all your hard work on this program.  It is fantastic.  I just found it yesterday and have already made some fairly complex objects on it.

whitequark, 2016-12-16:
OK, "Turntable" is a good user-visible name.

whitequark, 2019-04-24:
@dynamodan Unfortunately I think I can't merge this yet :( There's a bug: if you use View → Nearest Ortho View (or similar commands), then you can get into a state where the Z axis points somewhere completely else than up, and you can't do anything about it.

Symbian9, 2019-04-27:
> here's a bug: if you use View → Nearest Ortho View (or similar commands), then you can get into a state where the Z axis points somewhere completely else than up, and you can't do anything about it.

FTR, LeoCAD has good implementation on view control in with Z-Up and Perspective/Ortho switching.

- https://github.com/leozide/leocad
  - https://github.com/leozide/leocad/releases/tag/continuous

Maybe, we need few view control modes:

- *Default* (currently availabe in SolveSpace)
- *Turntable* (as proposed above, **but with disabled "View → Nearest Ortho View"**)
- other modes.

whitequark, 2019-04-27:
> FTR, LeoCAD has good implementation on view control with Z-Up, Perspective/Ortho switching and much more:

If you want to be helpful, you can write the code for this feature, of course.

dynamodan, 2019-04-27:
@Symbian9 instead of disabling the nearest ortho view, I'm considering simply modifing the nearest N view functions so that, when in turntable mode, Z remains correct.

Another way to understand "turntable" mode is to think of the horizon always staying level and not banking as it would in a flight simulator when the airplane rolls.  It should allow the camera to pitch and yaw, but restrict roll.  I explore the bug in this video:

https://youtu.be/siLoaXG3nM4

whitequark, 2019-05-01:
> instead of disabling the nearest ortho view, I'm considering simply modifing the nearest N view functions so that, when in turntable mode, Z remains correct.

This sounds like the right approach to me. AnimateOnto will also need to be fixed.

whitequark, 2019-05-23:
Merged in e67f967.

closed 2019-05-23 by whitequark