solvespace/solvespace #124: macOS: Export 2D view file selection inconsistent if file extensions are hidden (default)

michthom, 2016-11-28:
When returning from the GetSaveFile function in cocoamain.mm, the modal dialog to select a filename defaults to adding .pdf even if the dropdown menu is changed to DXF for example, **if** 'hide extensions' is active (default Finder behaviour these days).

To reproduce:
Check that 'Show all filename extensions' is UNCHECKED in Finder Preferences/Advanced tab. 
File: Export 2D view
Leave file name as 'untitled' (default) and File Format as 'PDF' - default.
Click OK

Result - PDF as expected.

File: Export 2D view
Leave file name as 'untitled' (default)
Change File Format to DXF
Click OK
Receive warning about overwriting 'untitled.pdf'
Click Replace

Result - PDF - not expected - assumed the file written would be .dxf in DXF format

File: Export 2D view
Change file name to 'untitled.dxf'
Change File Format to DXF
Click OK

Result - DXF file saved.

Does the URL path returned by fileSystemRepresentationWithPath not get the extension added (if extensions are hidden), which throws VectorFileWriter::ForFile into default PDF mode?

Workaround: Enable file extensions in Finder Preferences?
Workaround: Enable canSelectHiddenExtension in the NSSavePanel?

whitequark, 2016-11-28:
*sigh*

michthom, 2016-11-28:
Not really your bug to fix, more a macOS oddity? I'd assumed the URL passed back should include the extension even if the Finder isn't displaying it - but from the behaviour (I haven't traced the code properly) I'm assuming you get the bare filename and the directory only?

whitequark, 2016-11-28:
I have no idea, I'll load it up in a debugger...

whitequark, 2016-11-29:
This is only on macOS 10.10+ so you're on your own. (The process of upgrading a VM is very obnoxious.)

whitequark, 2016-11-29:
btw [NSSavePanel canSelectHiddenExtension] is only available on 10.10+ and I think we currently use a deployment target of 10.7.

whitequark, 2018-07-18:
Fixed in master.

closed 2018-07-18 by whitequark