solvespace/solvespace #127: Linking solvespace-cli fails

mlen, 2016-12-02:
It seems like the command is missing the `-lcairo` flag
```
/usr/bin/x86_64-pc-linux-gnu-g++   -march=broadwell -O2 -pipe -Wall -Wextra -Wno-unused-parameter -Werror=switch  -Wl,--as-needed -Wl,-O1 -Wl,--as-needed CMakeFiles/solvespace-cli.dir/platform/climain.cpp.o  -o ../bin/solvespace-cli -rdynamic ../bin/libsolvespace-core.a ../bin/libsolvespace-headless.a ../bin/libsolvespace-core.a ../bin/libdxfrw.a -lz -lpng -lfreetype

../bin/libsolvespace-headless.a(headless.cpp.o): In function `SolveSpace::PaintGraphics()':
headless.cpp:(.text+0x6b2): undefined reference to `cairo_format_stride_for_width'
headless.cpp:(.text+0x70b): undefined reference to `cairo_image_surface_create_for_data'
headless.cpp:(.text+0x716): undefined reference to `cairo_create'
headless.cpp:(.text+0xa60): undefined reference to `cairo_surface_destroy'
headless.cpp:(.text+0xa68): undefined reference to `cairo_destroy'
../bin/libsolvespace-headless.a(rendercairo.cpp.o): In function `SolveSpace::CairoRenderer::OutputStart()':
rendercairo.cpp:(.text+0x13): undefined reference to `cairo_save'
rendercairo.cpp:(.text+0x72): undefined reference to `cairo_rectangle'
rendercairo.cpp:(.text+0xc9): undefined reference to `cairo_set_source_rgba'
rendercairo.cpp:(.text+0xd5): undefined reference to `cairo_fill'
rendercairo.cpp:(.text+0x11d): undefined reference to `cairo_translate'
rendercairo.cpp:(.text+0x135): undefined reference to `cairo_translate'
rendercairo.cpp:(.text+0x146): undefined reference to `cairo_set_line_join'
../bin/libsolvespace-headless.a(rendercairo.cpp.o): In function `SolveSpace::CairoRenderer::FinishPath()':
rendercairo.cpp:(.text+0x1fc): undefined reference to `cairo_has_current_point'
../bin/libsolvespace-headless.a(rendercairo.cpp.o): In function `SolveSpace::CairoRenderer::OutputEnd()':
rendercairo.cpp:(.text+0x231): undefined reference to `cairo_restore'
rendercairo.cpp:(.text+0x23d): undefined reference to `cairo_get_target'
../bin/libsolvespace-headless.a(rendercairo.cpp.o): In function `SolveSpace::CairoRenderer::MoveTo(SolveSpace::Vector)':
rendercairo.cpp:(.text+0x278): undefined reference to `cairo_get_current_point'
rendercairo.cpp:(.text+0x284): undefined reference to `cairo_has_current_point'
rendercairo.cpp:(.text+0x2a8): undefined reference to `cairo_move_to'
../bin/libsolvespace-headless.a(rendercairo.cpp.o): In function `SolveSpace::CairoRenderer::OutputTriangle(SolveSpace::STriangle const&)':
rendercairo.cpp:(.text+0x36b): undefined reference to `cairo_set_source_rgba'
rendercairo.cpp:(.text+0x37c): undefined reference to `cairo_set_antialias'
rendercairo.cpp:(.text+0x392): undefined reference to `cairo_move_to'
rendercairo.cpp:(.text+0x3a8): undefined reference to `cairo_line_to'
rendercairo.cpp:(.text+0x3be): undefined reference to `cairo_line_to'
../bin/libsolvespace-headless.a(rendercairo.cpp.o): In function `SolveSpace::CairoRenderer::SelectStroke(SolveSpace::Canvas::hStroke)':
rendercairo.cpp:(.text+0x530): undefined reference to `cairo_set_line_width'
rendercairo.cpp:(.text+0x547): undefined reference to `cairo_set_dash'
rendercairo.cpp:(.text+0x5af): undefined reference to `cairo_set_source_rgba'
rendercairo.cpp:(.text+0x5ca): undefined reference to `cairo_set_antialias'
rendercairo.cpp:(.text+0x5f6): undefined reference to `cairo_set_antialias'
../bin/libsolvespace-headless.a(rendercairo.cpp.o): In function `SolveSpace::CairoRenderer::OutputBezier(SolveSpace::SBezier const&, SolveSpace::Canvas::hStroke)':
rendercairo.cpp:(.text+0x7ce): undefined reference to `cairo_arc'
rendercairo.cpp:(.text+0x82f): undefined reference to `cairo_curve_to'
rendercairo.cpp:(.text+0x841): undefined reference to `cairo_arc_negative'
rendercairo.cpp:(.text+0x87b): undefined reference to `cairo_line_to'
../bin/libsolvespace-headless.a(rendercairo.cpp.o): In function `SolveSpace::CairoRenderer::OutputStart()':
rendercairo.cpp:(.text+0x15f): undefined reference to `cairo_set_line_cap'
../bin/libsolvespace-headless.a(rendercairo.cpp.o): In function `SolveSpace::CairoRenderer::FinishPath()':
rendercairo.cpp:(.text+0x219): undefined reference to `cairo_stroke'
../bin/libsolvespace-headless.a(rendercairo.cpp.o): In function `SolveSpace::CairoRenderer::OutputEnd()':
rendercairo.cpp:(.text+0x246): undefined reference to `cairo_surface_flush'
../bin/libsolvespace-headless.a(rendercairo.cpp.o): In function `SolveSpace::CairoRenderer::OutputTriangle(SolveSpace::STriangle const&)':
rendercairo.cpp:(.text+0x3d0): undefined reference to `cairo_fill'
collect2: error: ld returned 1 exit status
```

whitequark, 2016-12-03:
That's odd, are you using a clean build directory?

mlen, 2016-12-03:
Yup, I build it using portage. It used to build few months ago, but it no longer builds using the same ebuild (below). I don't see anything obviously wrong in it though.
```
# Copyright 1999-2016 Gentoo Foundation
# Distributed under the terms of the GNU General Public License v2
# $Id$

EAPI=6

inherit cmake-utils git-r3

EGIT_REPO_URI="git://github.com/solvespace/${PN}.git https://github.com/solvespace/${PN}.git"
EGIT_SUBMODULES=("extlib/libdxfrw")
if [ "${PV}" != "9999" ]; then
	EGIT_COMMIT="${PV}"
fi

DESCRIPTION="Parametric 2d/3d CAD"
HOMEPAGE="http://solvespace.com"

LICENSE="GPL3"
SLOT="0"
KEYWORDS="~amd64 ~x86"

DEPEND="sys-libs/zlib
dev-libs/json-c
dev-cpp/gtkmm:2.4
dev-cpp/pangomm
media-libs/libpng
media-libs/freetype
media-libs/fontconfig
media-libs/glew
media-libs/glu
media-libs/mesa"

RDEPEND="${DEPEND}"

src_configure() {
	local mycmakeargs=("-DENABLE_TESTS=OFF")
	cmake-utils_src_configure
}
```

[Build log](https://gist.github.com/mlen/c17b5439d6065f6fc8871d850935d3f1)

whitequark, 2016-12-04:
Fixed in 6989a3d.

closed 2016-12-04 by whitequark