solvespace/solvespace #131: FindWhichToRemoveToFixJacobian can be very slow

whitequark, 2016-12-12:
See attached sketch--try adding a redundant constraint.
[constrainfreeze.zip](https://github.com/solvespace/solvespace/files/645378/constrainfreeze.zip)

Evil-Spirit, 2016-12-12:
@whitequark, we can limit number of constraints when FindWhichToRemoveToFixJacobian should be performed, if greater, we just report an error without concrete constraints.

whitequark, 2016-12-12:
@Evil-Spirit is there no faster way than just bruteforce?

Evil-Spirit, 2016-12-12:
@whitequark, Yes, we can be a little bit faster now (~3 times faster with the current example) if we just rewrite SolveBySubstitution for linear complexity instead of square. I actually done this for NotSolveSpace and this is huge speedup when we have a large amount of h,v,pc. The other two hot functions WriteJacobian and CalculateRand. The first can be performed once and used for many times, and the second can be low-level optimized or replaced in future by external linear system solver functions.

whitequark, 2016-12-12:
@Evil-Spirit OK, this sounds like something useful to be done. For now it is low priority though.

Evil-Spirit, 2016-12-12:
@whitequark,
I've made attempt to make some optimization, mostly because we have to exclude SolveBySubstitution because of #208. But this still o(n^4)....(((((

Evil-Spirit, 2016-12-13:
@whitequark 
One more imporvement.
But we can't avoid SolveBySubsitution because constrainfreeze.zip without it and even with optimizations is dead slow (~60sec). I think we need to analyze redundant h/v/pc not using RankTest, just comparing constraints.

Evil-Spirit, 2016-12-13:
@whitequark 
https://github.com/Evil-Spirit/solvespace-master/commit/6584093cb3ce23ca5fb87fe39877d5448fe854ea

whitequark, 2017-01-14:
Doesn't apply on master anymore.

Evil-Spirit, 2017-01-14:
@whitequark, OK I am rebasing

Evil-Spirit, 2017-01-14:
https://github.com/Evil-Spirit/solvespace-master/commits/master-131

Evil-Spirit, 2017-01-14:
@whitequark, but this is works bad after we add forceDofCheck. try to comment forceDofCheck inside FindWhichToRemoveToFixJacobian to allow SolveBySubstitution run. It will be fast for constraintfreeze.slvs. But this will not consider some cases with double-hvpc

whitequark, 2017-01-14:
@Evil-Spirit how to fix this?

Evil-Spirit, 2017-01-14:
@whitequark, For now I only can suggest to limit FindWhichToRemoveToFixJacobian by time. For constraintfreeze it's over 600 equations if we don't appy solvebysubstitution, so we can just give up until we involve sparse-matix algorithms. The second way - is to find double constraints like hvpc

whitequark, 2017-01-14:
@Evil-Spirit ok. is there any benefit from your updated patch?

Evil-Spirit, 2017-01-14:
@whitequark, only if we choose "finding the same multiple hvpc". if we bound by time, there is no point.
only benefit is from loop unrolling, but this is something like x2-x3

whitequark, 2017-01-14:
@Evil-Spirit ok. let's add a 1000ms time limit, and indicate in the text window if it was reached.

Evil-Spirit, 2017-01-14:
@whitequark, acc.

Evil-Spirit, 2017-01-14:
@whitequark, implemented here

commit 61570844 (2020-09-16): Fix #131. Prevent UI freeze by having a timeout when finding which constraints can be removed to fix jacobian.
closed 2020-09-16 by phkahler