solvespace/solvespace #132: Accept comma as decimal separator (enhancement)

Contraptionize, 2016-12-12:
It would be nice to be able to enter dimensions on international keyboards that feature a comma on the numpad instead of a period / full stop as decimal separator. Currently this causes the Error message "Not a valid number or expression: "1,23".

The decimal point is currently hardcoded in https://github.com/solvespace/solvespace/blob/master/src/expr.cpp#L706

It might suffice to replace it just after https://github.com/solvespace/solvespace/blob/master/src/expr.cpp#L683
`if( c == ',') input[inputPos] = '.';`

although there likeley are better solutions to do this.

There could be an option to enable this behavior to avoid unexpected behaviour for people trying to use the comma as thousands-separator (e.g. 1,000.00 - one thousand), although this does is not supported anyway.

Keywords: internationalization, i18n, localization, L10n

whitequark, 2016-12-12:
I don't see how this will work once functions are implemented. E.g.: think of `atan2(1,0)`.

Contraptionize, 2016-12-12:
Yes, sorry, I have not really looked at the parser code and did not think of comma as argument separator.

whitequark, 2016-12-12:
Note that I don't reject this so far, l10n is a valid concern. What do other applications do? Is this why Excel uses `;` as a separator? Do other numeric processors just force you to input `,`?

whitequark, 2016-12-13:
Regarding thousands-separator, the way endorsed by SI/ISO 31-0 is to separate them with spaces, not commas; we can use _ in place of spaces, I think.

whitequark, 2016-12-13:
Regarding numpad-dot/comma, I think we can just detect that key and translate it to a dot.

whitequark, 2017-01-02:
After some consideration, and [asking](https://twitter.com/whitequark/status/808630690166099968) people using non-US keyboard layouts, I have decided that this will get handled in two ways:
  - [ ] `_` will be accepted and ignored when parsing expressions;
  - [ ] numpad `.`/`,` will be always translated to `.`.

whitequark, 2017-01-02:
Fixed in master (not for Cocoa though since I think it doesn't expose the numpad keycodes in any way?)

closed 2017-01-02 by whitequark