solvespace/solvespace #137: Does this support 3D woodworking design and gcode generating?

codepython, 2016-12-19:


whitequark, 2016-12-19:
> 3D woodworking design

What features exactly are you looking for?

> gcode generating

You can generate 2.5D g-code but there is no support for a true CAM.

closed 2016-12-21 by whitequark

codepython, 2016-12-23:
what is the use of a 2.5d g-code?  is it for simulating?

whitequark, 2016-12-23:
You could e.g. create cutouts from sheet material on a CNC mill, say a sign or something.

Harvie, 2019-05-20:
It would be really cool if there was option to produce real 3D toolpaths, like real 3D CAM would do!
Is there any will to implement this?

whitequark, 2019-05-20:
It's an enormous challenge to do well (especially if you want HSM), and there's not much impetus to do an open-source CAM now that Fusion360 is free. So don't count on it...

Harvie, 2019-05-22:
> especially if you want HSM

High speed manufacturing is not crucial.

> now that Fusion360 is free

LOL. So why even bother with solvespace CAD, when Fusion360 is free?

It's not free opensource and does not run on Linux.
Also you can't be sure that it will be free of charge forever...

whitequark, 2019-05-22:
> LOL. So why even bother with solvespace CAD, when Fusion360 is free?

Yeah. A bunch of people I know dropped SolveSpace in favor of F360. That's how it goes...

altaic, 2019-05-22:
CAM is infinitely hard because it changes every time there’s a new process. Want to 3d print— Slic3r or Cura. Subtractive machining? Mach 3 (or 4 or whatever it is now). CAM is so out of scope here. Use a combination of tools to meet your workflow. There are an abundance of open source options.

Harvie, 2019-05-22:
> There are an abundance of open source options.

This is certainly not true. Show me good 3D CAM for Linux.
And don't even let me started with integrated 3D CAD/CAM for Linux.

whitequark, 2019-05-22:
Yeah, there's only really HeeksCNC and... it's kind of hard to use.