solvespace/solvespace #141: New rectangle. The dragged point are hovered.

ryannining, 2016-12-25:
### System information

SolveSpace version: (3.0~9148d0cb.)

Operating system: (e.g. Ubuntu testing)

### Expected behavior
The dragged point should not hovered, just like new line segment.

What should have happened?
The dragged point should not hovered, just like new line segment.

### Actual behavior
The dragged point constantly hovered.

What actually happened?
The dragged point constantly hovered.

### Additional information
If the DRAGGING_NEW_POINT the point is not hovered, then i can add ConstrainPointByHovered(pending.point); on mouseleftdown so the rectangle is constrained with other entity that hovered (not it self). If i add this anyway it will cause solver problem.

I have use this feature on my old version (sorry havent update on git). It work nicely.

For bugs, please attach a savefile that shows the problematic behavior.
You can attach `.slvs` files by archiving them into a `.zip` first.

ryannining, 2016-12-25:
Looks like the probllem is because this code is different:
old:
`        d = e->GetDistance(mp);
        if(d < SELECTION_RADIUS && d < dmin) {
            s = {};
            s.entity = e->h;
            dmin = d;
        }`

new:
`if(canvas.Pick([&]{ e.Draw(Entity::DrawAs::DEFAULT, &canvas); })) {
            s = {};
            s.entity = e.h;
        }`

something to do with min distance ?

Evil-Spirit, 2017-01-09:
@whitequark, fixed here. +improvements.

whitequark, 2017-01-11:
Fixed in master.

closed 2017-01-11 by whitequark

Evil-Spirit, 2017-01-11:
@whitequark,
what about https://github.com/solvespace/solvespace/commit/31296333539d6ce5f27220aba7363c2f26dfc1b2?
This is very annoying for me.

whitequark, 2017-01-11:
@Evil-Spirit I haven't noticed that one... you just linked one commit here.

Evil-Spirit, 2017-01-11:
@whitequark, Oh, sorry, usually I make branch for every issue and put some other improvements realated to it.

whitequark, 2017-01-11:
Merged in master too.