solvespace/solvespace #143: Export 2D View as Blueprint

Symbian9, 2016-12-28:
### Expected behavior

Pick checkbox "export 2d View as  Blueprint" for getting 4 views: side - face - back - isometric
[image]
* Blender addon that describe needed - https://github.com/Pullusb/SB_auto-blueprint

### Actual behavior

Currently for making expected future need manually create external slvs, and import 4 times one needed model, then also need place them correctly for 2D view export in 3D space. It's very hurd and make cause memory for complex models.

### Additional information

Blueprints widely used and very popular method for representation of 3D objects in static image.

whitequark, 2016-12-28:
Depends on #80.

ruevs, 2021-11-25:
https://solvespace.com/forum.pl?action=viewthread&parent=3996&tt=1636240197

Symbian9, 2021-11-26:
> https://solvespace.com/forum.pl?action=viewthread&parent=3996&tt=1636240197

To be clear, it is just set of drawing templates not related directly for auto generating blueprint (as I requested).

See also further discussion on drawing templates with its author:

- https://www.reddit.com/r/SolveSpace/comments/qocc3h/2d_drawings_in_solvespace/