solvespace/solvespace #145: Can't import DWG (public from Airbus site)

Symbian9, 2016-12-31:
### System information

SolveSpace version:  https://github.com/solvespace/solvespace/releases/tag/v2.3

Operating system: run EXE under Wine on PuppyLinux

### Expected behavior

Open 3views drawing from public DWG file.

### Actual behavior

SolveSpace try import, but can't continue operation (too long and not self stopped)

### Additional information

Test case:
* http://www.airbus.com/support/maintenance-engineering/technical-data/autocad-3-view-aircraft-drawings/
  * http://www.airbus.com/fileadmin/media_gallery/files/tech_data/AUTOCAD/Airbus-A350-1000.zip

Evil-Spirit, 2016-12-31:
@Symbian9, this is because of auto-constraining of an imported primitives (H/V/PC). We should made option to switch it off.

Evil-Spirit, 2016-12-31:
@Symbian9, this file is just not loading by libDXFrw, I have resaved it (with scale factor) and this is loading now.

whitequark, 2016-12-31:
The DWG support is on a best-effort basis. The format is undocumented and reverse-engineering it is a huge waste of time.

closed 2016-12-31 by whitequark