solvespace/solvespace #147: Circle intersections do not split properly

pretentiousgit, 2017-01-01:
### System information

SolveSpace version: 2.3

Operating system: Mac OSX El Cap 10.11.4

### Expected behavior

Select two circles and click "split lines/curves where they intersect"

Expected results might be:
-- four arcs, with two points where they intersect
-- Two arcs joined at end points with inner lines removed

### Actual behavior

Single data point connecting two circles appear. 
[image]

### Additional information

For bugs, please attach a savefile that shows the problematic behavior.
You can attach `.slvs` files by archiving them into a `.zip` first.

[circleBug.slvs.zip](https://github.com/solvespace/solvespace/files/680213/circleBug.slvs.zip)

ohsix, 2017-01-01:
tried this on master and it DTRT, circles look finer here though

in configuration what do you have chord tolerance and max piecewise linear segments set to?

(not a developer, just saw something i could try)

pretentiousgit, 2017-01-01:
Just have out of the box settings - haven't messed with things at all. Not sure if it's a valid bug or if I missed a step in the tutorial/the intent.

Evil-Spirit, 2017-01-02:
Also, this sould work with the list of entities

whitequark, 2017-03-12:
I think we should just document that one may need to use the tool multiple times, if several intersections are desired.

Evil-Spirit, 2017-03-12:
@whitequark, actually there is no problem to implement this.

whitequark, 2017-03-12:
@Evil-Spirit OK, if it's easy please write a patch