solvespace/solvespace #148: Dragging points is extremely slow after commit d99a133

whitequark, 2017-01-02:
E.g. try dragging a rectangle. While the mouse is moving no entities will be drawn correctly.

@Evil-Spirit please fix this ASAP

Evil-Spirit, 2017-01-02:
@whitequark, please, provide more information

whitequark, 2017-01-02:
@Evil-Spirit I have this instead of smooth dragging:
[image]

Evil-Spirit, 2017-01-02:
@whitequark, hm. I have no such problem on Windows.
Do you have this after Rect creation and dragging it again?

whitequark, 2017-01-02:
Only during initial creation of rect and line. All other entities are fine.

Evil-Spirit, 2017-01-02:
I have artifacts on initial dragging, but this is very old, it appears after we introduced "havePainted"

Evil-Spirit, 2017-01-02:
@whitequark 
I have fixed same-looking issue on windows. Can you try it for linux?

whitequark, 2017-01-02:
@Evil-Spirit No, same thing.

whitequark, 2017-01-02:
I think maybe you aren't marking a group as dirty sometimes.

Evil-Spirit, 2017-01-02:
@whitequark, Is InvalidateGraphics() actrually redraws all the window on linux?

whitequark, 2017-01-02:
Here's the code:
```
void InvalidateGraphics(void) {
    GW->get_widget().queue_draw();
}

void PaintGraphics(void) {
    GW->get_widget().queue_draw();
    /* Process animation */
    Glib::MainContext::get_default()->iteration(false);
}
```
InvalidateGraphics schedules a redraw, PaintGraphics schedules a redraw and immediately runs the message handling loop (just like WndProc) until all messages are processed.

Can you reproduce this in a VM?

Evil-Spirit, 2017-01-02:
@whitequark, can you turn back 
mouse.cpp:1391
```
        SS.MarkGroupDirty(c->group);
        SS.GenerateAll();
```
Probably PaintGraphics runs before GenerateAll

whitequark, 2017-01-02:
@Evil-Spirit That doesn't help. But I also found this only happens on GTK 2. So maybe this just means it's time to drop GTK 2 from SolveSpace 3, and fix any remaining bugs.

whitequark, 2017-01-02:
Solved by killing GTK2 and removing various horrible hacks also.

whitequark, 2017-01-02:
> I have artifacts on initial dragging, but this is very old, it appears after we introduced "havePainted"

Let's fix that too, what sort of artifact is that?

Evil-Spirit, 2017-01-02:
When dragging new rect one of the line not matched in vertex.  This is because redrawing before solving.  This happens inside hit test function.

whitequark, 2017-01-02:
@Evil-Spirit There must be some other error, because with both of your fixes above, dragging of new rect still has the same issue on Linux.

whitequark, 2017-01-03:
@Evil-Spirit I've merged your Windows fix.

Evil-Spirit, 2017-01-03:
@whitequark, more fixes

Evil-Spirit, 2017-01-03:
@whitequark, problem with mouse-move hovering happens again(viewport is not redrawing while moving over, just when mouse move stopped)

whitequark, 2017-01-03:
This fix you added makes no sense.

Evil-Spirit, 2017-01-04:
@whitequark,
We have eliminated only PaintGraphics() call inside HitTestMakeSelection, so there is no place where PaintGraphics worked while mouse move events present in queue and takes a lot of time.

Evil-Spirit, 2017-01-04:
@whitequark,
 This is only windows problem

whitequark, 2017-01-04:
Then I will revert that commit, because your fix makes no sense, and after it PaintGraphics/InvalidateGraphics don't do what they claim to.

Evil-Spirit, 2017-01-04:
@whitequark, Can you explain what does mean "don't do what they claim to"?
InvalidateGraphics before this just posts WM_PAINT message, PaintGraphics redraws immediately.
The things have worked before this way also works this way now (if WM_PAINT doesn't drew viewport it will be drawn using SS.later.paintGraphics, but never before GenerateAll as it can be done now)

whitequark, 2017-01-04:
And on GTK/Cocoa PaintGraphics redraws immediately. Any fix that makes the platforms diverge isn't the right fix.

Evil-Spirit, 2017-01-04:
@whitequark, OK, changed this

whitequark, 2017-01-05:
@Evil-Spirit That's much better! Let me see if this applies to

whitequark, 2017-01-12:
@Evil-Spirit Actually, I think your fix will create another problem. Try it on any case where hit testing is really slow.

Evil-Spirit, 2017-01-12:
@whitequark,
I have tested it for big-big-big-woodworking-asm.slvs(where HitTestMakeSelection really slow) and there is no problem for me (with using https://github.com/Evil-Spirit/solvespace-master/commit/6cdb9a3fd38e79706bf70b896f13e052f0df6890)

whitequark, 2017-02-06:
I've figured out the problem. It has nothing to do with invalidation. Rather, it is because of ScheduleGenerateAll that never actually performs the generation if the queue is full of mouse move events (at least on Linux).

Evil-Spirit, 2017-02-06:
@whitequark, Good. So how to fix this?

whitequark, 2017-02-06:
@Evil-Spirit Not sure. Maybe we need to rewrite the message handling loop on Linux to make it look like it does on Windows.

whitequark, 2017-02-08:
Fixed in master.

closed 2017-02-08 by whitequark

Cimera42, 2020-05-12:
> 
> 
> @Evil-Spirit I have this instead of smooth dragging:
> [image]

I'm getting this exact issue on Windows when building the current master branch (e84fd46). I've tried building on Linux with MinGW, and building on Windows with Visual Studio and MinGW.
I remember building back when Helix was added, and was getting the same issue.

Would love to know if anyone else is encountering this issue, and potential fixes. I've looked into the original fixes from above, but the file architecture has changed a lot since then.

Cimera42, 2020-05-12:
Although I don't know the impacts of doing so, I've been able to get it working properly by removing the [`Invalidate()` from `mouse.cpp`](https://github.com/solvespace/solvespace/blob/wip/src/mouse.cpp#L308).

```
case Pending::DRAGGING_NEW_POINT:
    UpdateDraggedPoint(pending.point, x, y);
    HitTestMakeSelection(mp);
    SS.MarkGroupDirtyByEntity(pending.point);
    orig.mouse = mp;
    //Invalidate();
    break;
```

whitequark, 2020-05-15:
Thanks for the investigation. The fix doesn't look right to me in general, but the fact that it helps for your specific case is still useful to know. Please open a new issue to track this problem; I was under the impression that it's fixed, but looks like it either never was and wasn't detected by my tests, or it got broken somewhere since 2017.