solvespace/solvespace #151: Font texture update not working

Evil-Spirit, 2017-01-09:
### System information
Windows, OpenGL2
### Expected behavior
Font texture should be updated for each OpenGL context.
### Actual behavior
Font texture updated only for context where new character appears
### Additional information
For native windows OpenGL there are possible to use the same context for different windows, or share context resources by using wglShareLists. Unfortunately, this approach can't work with ANGLE, so we have to support two contexts and update font texture for each of them.

whitequark, 2017-01-09:
yes, I also noticed this bug, please fix it

Evil-Spirit, 2017-01-09:
@whitequark,
I have no idea how to make this not using some hacking.
1. The best solution I know is perfrom versioning of the font texture and store it inside pixmapCache.
2. Make copy of the font for each Canvas instance

whitequark, 2017-01-09:
@Evil-Spirit 2nd sounds OK to me

Evil-Spirit, 2017-01-09:
@whitequark, OK!? ok :)

Evil-Spirit, 2017-01-10:
@whitequark, ready

whitequark, 2017-01-11:
@Evil-Spirit This crashes tests. Why do you not run tests?

whitequark, 2017-01-11:
Fixed in master.

closed 2017-01-11 by whitequark

Evil-Spirit, 2017-01-11:
@whitequark, I have different build directory, so I have to run tests separately and sometimes forget to do it. Sorry, I just need to reconfigure my building env to build inside repo dir.

whitequark, 2017-01-11:
@Evil-Spirit OK