solvespace/solvespace #153: Crash with Unmatched regex

Evil-Spirit, 2017-01-12:
### System information
Appears after adding translations.
Windows 7 x64, MS VS 2013 compiler

### Expected behavior

Should work

### Actual behavior

Crash on running tests and running SolveSpace.exe

### Additional information

File E:\SolveSpaceWork\solvespace_dxf\src\resource.cpp, line 488, function SolveSpace::ASCIIReader::ExpectRegex:
Assertion 'Unmatched regex' failed: ((TryRegex(re, m)) == false).

whitequark, 2017-01-12:
This seems like an MSVC-related bug, or maybe \r\n issue. Can you look?

Evil-Spirit, 2017-01-12:
@whitequark, OK

lomatus, 2017-01-13:
Vote +1
Appears after adding translations.
Windows 7 x64/x86, MS VS 2015 compiler

whitequark, 2017-01-14:
I cannot reproduce this one myself with either MSVS2013 or MSVS2015, weird.

@lomatus would that be a zh_CN translation? I am very interested in that one.

whitequark, 2017-01-14:
Should be fixed in master.

closed 2017-01-14 by whitequark

whitequark, 2017-01-14:
@Evil-Spirit btw there is an unmerged ru_RU translation already, don't bother

Evil-Spirit, 2017-01-14:
@whitequark, OK