solvespace/solvespace #154: color selection is confusing / buggy

wpwrak, 2017-01-12:
### System information

SolveSpace at commit 7dbbd75969b5f7c26e97ffe3e8f9da67fb3cfca4
(i.e., before the gtkmm-3.22 requirement, since Ubuntu is still at 3.20)

### Actual behavior

Color selection is confusing:
1) it is hard to distinguish the color selection overlay from the dialog in the background.
  E.g., the checkboxes on the left, the "mesh" on the right, etc. 
2) the logic of color selection is hard to guess and seems inconsistent.

### Expected behavior

I would expect that left-clicking on a palette item selects that color, that the hue (?) can be dragged, etc.
Instead, right-clicking a palette item sometimes selects that color, but that also seems to vary with context.

whitequark, 2017-01-12:
> I would expect that left-clicking on a palette item selects that color, that the hue (?) can be dragged, etc.

Yeah that's how it is supposed to work.

whitequark, 2017-01-17:
Fixed in master.

closed 2017-01-17 by whitequark