solvespace/solvespace #155: Build broken on Ubuntu 16.04

dynamodan, 2017-01-13:
### System information

SolveSpace version: latest from git, i.e. commit 022d012a4454952f45

Operating system: Ubuntu 16.04

### Expected behavior

Check out from git, follow compilation steps for Linux, run

### Actual behavior

Got this when running cmake:

$ cmake .. -DENABLE_TESTS=OFF
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using in-tree libdxfrw
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Looking for backtrace
-- Looking for backtrace - found
-- backtrace facility detected in default set of libraries
-- Found Backtrace: /usr/include  
-- Found SPACEWARE: /usr/lib/libspnav.so  
-- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found version "1.2.8") 
-- Found PNG: /usr/lib/x86_64-linux-gnu/libpng.so (found version "1.2.54") 
-- Found Freetype: /usr/lib/x86_64-linux-gnu/libfreetype.so  
-- Checking for module 'cairo'
--   Found cairo, version 1.14.6
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so  
-- Checking for module 'fontconfig'
--   Found fontconfig, version 2.11.94
-- Checking for module 'json-c'
--   Found json-c, version 0.11.99
-- Checking for modules 'gtkmm-3.0>=3.22;pangomm-1.4;x11'
--   
CMake Error at /usr/share/cmake-3.5/Modules/FindPkgConfig.cmake:367 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPkgConfig.cmake:532 (_pkg_check_modules_internal)
  CMakeLists.txt:194 (pkg_check_modules)

-- Configuring incomplete, errors occurred!

### Additional information

Ubuntu 16.04's latest GTK3 is at version 3.18.0-1 . Not until you get up to Ubuntu Zesty (17.04?) do you get a GTK3 at 3.22.  (This hasn't been released yet.)

If I relax the GTK3 version in CMakeLists.txt to require 3.18, then it's the set_use_es(true) lines in gtkmain.cpp that trip up compilation.  If I comment out these lines, then the compilation succeeds but all I get upon launch is grey windows with a message "OpenGL context creation failed".  I have mesa-utils-extra install which provides es2gears, which runs fine.  Any advice?  Thanks in advance.

whitequark, 2017-01-13:
This is pretty much working as intended. 3.0 does not (yet) have major new features and is not particularly close to release. I expect this will not be a problem as Debian Stretch releases with 3.22.

whitequark, 2017-01-13:
Actually, nevermind, there's an easy fix and I will apply it soon.

wpwrak, 2017-01-13:
For what it's worth, Ubuntu 16.10 (Yakkety Yak) gets a bit closer, gtkmm 3.20.
I found that with this, I can build SolveSpace at commit 7dbbd75969b5f7c26e97ffe3e8f9da67fb3cfca4 (from Jan 2, 2017).
If the requirement could be lowered at little, that would of course be even better :)

dynamodan, 2017-01-13:
I can build that commit on Ubuntu 16.04 as well.  It seems the Open GL ES stuff apparently isn't supported on GTK3 until 3.22.

Symbian9, 2017-01-13:
@whitequark, add "Linux" tag to this issue.

whitequark, 2017-01-14:
Fixed in master.

closed 2017-01-14 by whitequark

drbobbob, 2017-01-14:
The build works now (29e43e67eaea5c7f95fc0943a6728583cb9efbcf) but running does not work on my Ubuntu 16.04 system.  I get the following error and the screen shows "OpenGL context creation failed"

```
drbobbob@ray-lappy-lenovo:~/workspace/solvespace/build2$ bin/solvespace

(solvespace:2588): GLib-GObject-CRITICAL **: g_value_take_object: assertion 'G_IS_OBJECT (v_object)' failed

(solvespace:2588): GLib-GObject-CRITICAL **: g_value_take_object: assertion 'G_IS_OBJECT (v_object)' failed
SolveSpace!
```

whitequark, 2017-01-14:
Haven't pushed everything, try again.

drbobbob, 2017-01-14:
Works with fba88859e1646e821e3b61700a02fe8c37ba324d.  Thanks for the quick turnaround.

dynamodan, 2017-01-14:
@whitequark Thanks a bunch!  I had encountered the same issue as @drbobbob with the G_IS_OBJECT assertion bug, but everything's working now.

Symbian9, 2017-01-17:
@drbobbob, @dynamodan, are you has working PPA's for this builds? Add links to them in related issue discussion
* https://github.com/solvespace/solvespace/issues/142

dynamodan, 2017-01-17:
@Symbian9 No, I am not aware of a PPA for this build.  The only way I get solvespace is by building from source.