solvespace/solvespace #157: better support for dependency analysis

wpwrak, 2017-01-13:
Sometimes, it can be difficult to determine which constraints are in place, and to what objects they are applied. What would help is to have slightly more tools for exploring how items in a sketch are related, e.g.:

- when selecting a point, line, etc., get a list of (clickable) constraints in the text window,
- when selecting a constraint, get a list of (clickable) items it applies to (in addition to the highlighting),
- and maybe also the possibility of labeling (naming) things, even if only temporarily (i.e., without saving), so that one doesn't have to identify the same item over and over again.

Evil-Spirit, 2017-01-16:
May be something like this can be useful too.
[image]

whitequark, 2017-01-16:
@Evil-Spirit How does that work?

Evil-Spirit, 2017-01-16:
@whitequark,
We draw stippled line between two points (get it using GetReferencePoints) when drawing hovered constraint.

whitequark, 2017-01-16:
@Evil-Spirit Excellent idea.

wpwrak, 2017-01-16:
The circles on hover would be great for better visualization.

However, I'm also looking for making selection unambiguous. E.g., if I want to do something with a point that I've identified through a constraint, I could unambiguously select it if the dependencies are shown (and selectable) in the text window. However, if it's only in the graphics window, clicking on that point may select something else, especially if the problem I'm trying to solve is some issue with points not quite coinciding in the model, while conceptually they should.

I'm not entirely sure this would be a real-life issue. I noticed the impossibility of reliably tracking ("debugging", really) such things when I had doubts about the structure of just such a situation. I.e., I couldn't quite tell whether that corner where everything met had just one point or multiple overlapping ones.

Evil-Spirit, 2017-01-16:
@whitequark, I think we can easily list all constraints liked to the selected entity

whitequark, 2017-01-17:
> when selecting a point, line, etc., get a list of (clickable) constraints in the text window,
done

> when selecting a constraint, get a list of (clickable) items it applies to (in addition to the highlighting),
done

> and maybe also the possibility of labeling (naming) things, even if only temporarily (i.e., without saving), so that one doesn't have to identify the same item over and over again.
this is a fairly invasive change. if it is done, it will involve changing the file format. please use the newly implemented features, and if they are really not enough, reopen and i will implement this.

closed 2017-01-17 by whitequark

wpwrak, 2017-01-17:
Thanks a lot ! Now, when exploring the dependency graph, I find that a "back" button (like in Web browsers) would be really handy. I guess, next, I'll ask for "Forward", too :)

whitequark, 2017-01-17:
@wpwrak This isn't really implementable in a non-horrible way with the current text window code, so that will have to wait further.