solvespace/solvespace #159: Crash when deleting a group

whitequark, 2017-01-16:
Load [box.zip](https://github.com/solvespace/solvespace/files/708240/box.zip). Then delete group g006-sk-window. The crash results:
```
File /home/whitequark/Work/solvespace/src/dsc.h, line 334, function FindById:
Assertion 'Cannot find handle' failed: ((t != NULL) == false).
Backtrace:
 0: ./bin/solvespace(_ZN10SolveSpace14assert_failureEPKcjS1_S1_S1_+0x78) [0x5648a096e107]
 1: ./bin/solvespace(_ZN10SolveSpace6IdListINS_5GroupENS_6hGroupEE8FindByIdES2_+0x5a) [0x5648a0937dbc]
 2: ./bin/solvespace(_ZN10SolveSpace6Sketch8GetGroupENS_6hGroupE+0x20) [0x5648a0937450]
 3: ./bin/solvespace(_ZN10SolveSpace12SolveSpaceUI13GroupsInOrderENS_6hGroupES1_+0x3f) [0x5648a09bd2af]
 4: ./bin/solvespace(_ZN10SolveSpace12SolveSpaceUI11PruneGroupsENS_6hGroupE+0x3c) [0x5648a09bd382]
 5: ./bin/solvespace(_ZN10SolveSpace12SolveSpaceUI11GenerateAllENS0_8GenerateEbb+0x48f) [0x5648a09bdbab]
 6: ./bin/solvespace(_ZN10SolveSpace12SolveSpaceUI11GenerateAllENS0_8GenerateEbb+0x2d1) [0x5648a09bd9ed]
 7: ./bin/solvespace(_ZN10SolveSpace10TextWindow17ScreenDeleteGroupEij+0x97) [0x5648a0a44fb3]
 8: ./bin/solvespace(_ZN10SolveSpace10TextWindow10MouseEventEbbdd+0x2e1) [0x5648a0a4c391]
 9: ./bin/solvespace(_ZN10SolveSpace10TextWidget21on_button_press_eventEP15_GdkEventButton+0x9c) [0x5648a094ccaa]
10: /usr/lib/x86_64-linux-gnu/libgtkmm-3.0.so.1(_ZN3Gtk12Widget_Class27button_press_event_callbackEP10_GtkWidgetP15_GdkEventButton+0x54) [0x7f0be2a00a94]
11: /usr/lib/x86_64-linux-gnu/libgtk-3.so.0(+0x22fb11) [0x7f0be1601b11]
12: /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(+0x101a4) [0x7f0be0c911a4]
13: /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(g_signal_emit_valist+0x601) [0x7f0be0cab391]
14: /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(g_signal_emit+0x8f) [0x7f0be0cabfaf]
15: /usr/lib/x86_64-linux-gnu/libgtk-3.so.0(+0x37c3dc) [0x7f0be174e3dc]
16: /usr/lib/x86_64-linux-gnu/libgtk-3.so.0(+0x22cace) [0x7f0be15feace]
17: /usr/lib/x86_64-linux-gnu/libgtk-3.so.0(gtk_main_do_event+0x79e) [0x7f0be1600b5e]
18: /usr/lib/x86_64-linux-gnu/libgdk-3.so.0(+0x36bf5) [0x7f0be1111bf5]
19: /usr/lib/x86_64-linux-gnu/libgdk-3.so.0(+0x67be2) [0x7f0be1142be2]
20: /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x2a7) [0x7f0be09b77f7]
21: /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x4aa60) [0x7f0be09b7a60]
22: /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_loop_run+0xc2) [0x7f0be09b7d82]
23: /usr/lib/x86_64-linux-gnu/libgtk-3.so.0(gtk_main+0x85) [0x7f0be15ffcc5]
24: /usr/lib/x86_64-linux-gnu/libgtkmm-3.0.so.1(_ZN3Gtk4Main3runERNS_6WindowE+0x121) [0x7f0be299afb1]
25: ./bin/solvespace(main+0x53b) [0x5648a094a5eb]
26: /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf1) [0x7f0bdf20e2b1]
27: ./bin/solvespace(_start+0x2a) [0x5648a09306ca]
Aborted
```

This is a crash introduced in 022d012a4454952f454e33278af744c01f69667a, which I have reverted.

Evil-Spirit, 2017-01-16:
@whitequark 
```
@@ -57,6 +57,8 @@ bool SolveSpaceUI::PruneOrphans() {
 bool SolveSpaceUI::GroupsInOrder(hGroup before, hGroup after) {
     if(before.v == 0) return true;
     if(after.v  == 0) return true;
+    if(!GroupExists(before)) return false;
+    if(!GroupExists(after)) return false;
     int beforep = SK.GetGroup(before)->order;
     int afterp = SK.GetGroup(after)->order;
     if(beforep >= afterp) return false;
```

Evil-Spirit, 2017-01-16:
@whitequark, fixed here

whitequark, 2017-01-16:
Reapplied in master

closed 2017-01-16 by whitequark