solvespace/solvespace #160: center-of-gravity calculation

whitequark, 2017-01-17:
For design of rotating parts, it would be useful to have a centre-of-gravity calculation, with display of a point and its coordinates.

A parametric one would also be interesting. But, if placed into a group with a mesh, this creates a circular dependency. Perhaps it could be restricted to only be placed in sketch in workplane/3d groups.

Evil-Spirit, 2017-01-17:
@whitequark, What algorithm you can suggest? I know what we have to integrate by volume... I think there is only numeric method we can implement in simple way...

whitequark, 2017-01-17:
Sure, it can only be numeric, like volume and area calculations.

Evil-Spirit, 2017-01-17:
@whitequark, circular dependencies is the point of exisnace of the solver...

Evil-Spirit, 2017-01-17:
@whitequark, volume of an triangle mesh actually is not numerical but exact

whitequark, 2017-01-17:
But I don't think it will nicely work with the centre-of-gravity constraint because of discontinuities.

Evil-Spirit, 2017-01-17:
@whitequark, Yes, dymamic volume is quite complex task, but assembly is much simplier and actually can be implemented.

whitequark, 2017-01-17:
@Evil-Spirit let's start with the simplest thing, just non-parametric display, and then do the complex one if it's really useful.

Evil-Spirit, 2017-01-17:
@whitequark,
We can use regular 3d grid with fixed step (cubes) or Monte-Carlo approach for performing this kind of integraion.

whitequark, 2017-01-17:
@Evil-Spirit I trust you to make the right choice. I think a live display that updates when dragging is useful, so that probably should affect the decision.

Evil-Spirit, 2017-01-17:
@whitequark, No problems here. There is no need to perform numerical integration, we can exactly calculate center of mass of triangle mesh like we done it for volume! http://e-maxx.ru/algo/gravity_center

whitequark, 2017-01-17:
OK. How expensive is that?

Evil-Spirit, 2017-01-17:
@whitequark, exactly just like volume calculation.

Evil-Spirit, 2017-01-17:
@whitequark, implemented here
We can also change volume calculation to simplier one.

Evil-Spirit, 2017-01-17:
@whitequark, one more time

Evil-Spirit, 2017-01-17:
@whitequark, actually I have no idea how to test it.

whitequark, 2017-01-18:
@Evil-Spirit Hm, I want a different UI. Basically recompute it after every regeneration, show a bright dot in the center of the screen, and write the exact coordinates. This constraint is not really useful, without more in-depth solver support I'd have to get the center of mass to the center of rotation manually, which is fine, just make it more convenient.

Evil-Spirit, 2017-01-18:
@whitequark,
So, you don't need it actually to be as datum point. Just visualization with ability to see coordinates? Can we just introduce checkbox "show mass center of solid model" inside text window?

Evil-Spirit, 2017-01-18:
@whitequark,OK. What about correctness? Can you suggest samples/method of testing?

whitequark, 2017-01-18:
@Evil-Spirit I would run some quick checks with shapes where one can calculate it by hand. I've done that in high school but has since forgot everything, hope you haven't..

Evil-Spirit, 2017-01-18:
@whitequark, updated here

Evil-Spirit, 2017-01-18:
@whitequark, Now it is more useful
[image]

whitequark, 2017-01-19:
Merged

closed 2017-01-19 by whitequark