solvespace/solvespace #164: support reordering of groups

wpwrak, 2017-01-19:
When we have a group A and "child" groups B and C where B and C only depend on A (or things before A), then, in the group list, sequences A, B, C, and A, C, B both would satisfy the dependency requirements.

However, which sequence gets chosen depends on the order in which things were done during construction, and cannot be changed afterwards.

Since one may want to introduce a dependency/constraint between B and C, which is only possible from later to earlier group, it would be good if groups that have no conflicting dependencies could be moved/reordered in the group list.

AresAndy, 2022-05-15:
Any news on this? It has been 5y and this feature would be extremely useful

phkahler, 2022-05-15:
@AresAndy  no progress on this. Is your desire for this feature related to assemblies? I ask because I'd like to change the way assembly works so multiple linked files can exist in the same group - in your case if B and C were linked files, they would be in the same group and both move freely until constrained. But I make no promises on timeframes :-)

kbrock, 2022-05-16:
When I am exploring in SolveSpace, rather than copying a diagram I often create a few groups in an order: A, B, C, where a cube in C depends upon a point in B. There is only one file, no assemblies.

I later decide that it would work better if the order were A, C, B.
So I change the line in C to depend upon a point in A.
At this time, nothing in C depends upon B.

I would like to change the order of the groups to be A, C, B.

At least that is my use case.

---

Starting B's workplane perpendicular to A's workplane, and basing it upon a point in A, tends to alleviate much of the changes for me. Then extruding the shape in C upon a square in B tends to give me a lot of flexibility. But I end up with so many groups and inevitably I get the ordering wrong.

thanks again for the great product. This is the only CAD software that has really clicked for me.

AresAndy, 2022-05-24:
@phkahler Well, allowing groups to be sorted manually could unlock all sorts of possibilities (and weird things too I'm afraid, by side-effects), but the main and cool feature is that one could treat the groups tree as a _timeline_ so-to-speak. Eg. booleans would for instance apply only to the geoms existing up to it, anything further down the line remains untouched. Graphically:

- cube extrusion (**A**)
- cylinder rotation, intersect mode (**B**)
- other geom close to A (**C**)

In this scenario with said feature, **B** would not apply to **C** -even if it overlaps the cylinder- because it comes after in the timeline, whereas **A** is affected. It could also better highlight the order of operations, and perhaps help in the CAM process too