solvespace/solvespace #165: make arc-center relation "explorable"

wpwrak, 2017-01-22:
Now that we can follow the relation between items and constraints, it would be nice if one could do the same for arcs and their center point. (The same would apply for circles, and possibly other items with "special" points.)

Right now, the center point is shown in the text window just as "POINT", without indication that it is the center of an arc. It would be nice if one could 1) tell that it is special, e.g., by calling it "CENTER POINT" or such, or maybe displaying a center-of-arc constraint (if that makes sense), and 2) reach the point from the arc and vice versa, much like we can now reach items through the constraints that connect them.

wpwrak, 2017-01-22:
I forget an even more common case: the endpoints of lines.

Evil-Spirit, 2017-01-23:
@whitequark, implemented here

wpwrak, 2017-02-02:
Thanks ! Here's a fix for a little typo:
https://github.com/wpwrak/solvespace/commit/782a5bbbe61c0a1e5326d55218de33e6a7682f26

I also found this extra semicolon, which already existed before #165. I'm not sure if this is a safe change, though, or if something may depend on that typo:
https://github.com/wpwrak/solvespace/commit/a77cedbfd8dedd147fb59ebf9af936f4259ff31b

Evil-Spirit, 2017-02-02:
@wpwrak, thank you :) We've found this semicolon a long time ago, but I forgot to fix it.

wpwrak, 2017-02-02:
By the way, along the lines of explorability, how about something like this ? https://github.com/wpwrak/solvespace/commit/41e55b3d79f7d2df61fb9db0b494ff09a37d6add
(My ultimate goal in this quest of mine is to be able to perform the deletion in #182 without losing any groups, and this should bring me a little closer to being able to rearrange the dependencies.)

whitequark, 2017-03-11:
FWIW I haven't merged the patch because the UI work in it is horrible. It's completely inconsistent with the rest of text window stuff.

Evil-Spirit, 2017-03-23:
@whitequark, what do you mean?

ruevs, 2022-01-06:
I agree with @whitequark that the proposed implementation was too intrusive and inconsistent with the style of the UI code for the test window. Here https://github.com/solvespace/solvespace/pull/1182 is a clean implementation with minimal intrusiveness.

Whether the style of the UI code is nice or not is another question. I'm split on this...

Symbian9, 2022-01-07:
> I agree with `@whitequark`

@ruevs, just a reminder, `@whitequark` asked to not cast (mention/trigger) she in SolveSpace related threads anymore.

ruevs, 2022-01-07:
OK thanks.

PR #1182 (2022-01-08): Make points/vectors/normals shown in the Property Browser into active links
This makes all points and appropriate vectors/normals shown in the Property Browser window explorable/selectable and closes #165.

This is a clean implementation that follows the style of the UI code for
the text window. Previous unmerged attempts from the above issue are:
https://github.com/Evil-Spirit/solvespace-master/commit/eb3db32059bc5b3a4264aba7524ea1cd3e39b8f3
https://github.com/wpwrak/solvespace/commit/782a5bbbe61c0a1e5326d55218de33e6a7682f26
https://github.com/wpwrak/solvespace/commit/a77cedbfd8dedd147fb59ebf [... cut]
commit 122396c5 (2022-01-08): Make all points shown in the Property Browser into active links

This makes all points explorable/selectable and fixes #165 partially.

This is a clean implementation that follows the style of the UI code for
the text window. Previous unmerged attempts from the above issue are:
https://github.com/Evil-Spirit/solvespace-master/commit/eb3db32059bc5b3a4264aba7524ea1cd3e39b8f3
https://github.com/wpwrak/solvespace/commit/782a5bbbe61c0a1e5326d55218de33e6a7682f26
https://github.com/wpwrak/solvespace/commit/a77cedbfd8dedd147fb59ebf9af936f4259ff31b
https://github.com/wpwrak/solvespace/commit/41e55b3d79 [... cut]
commit 54f34fde (2022-01-08): Make vectors/normals shown in the Property Browser into active links

This makes appropriate vectors/normals explorable/selectable and closes #165.

This is a clean implementation that follows the style of the UI code for
the text window. Previous unmerged attempts from the above issue are:
https://github.com/Evil-Spirit/solvespace-master/commit/eb3db32059bc5b3a4264aba7524ea1cd3e39b8f3
https://github.com/wpwrak/solvespace/commit/782a5bbbe61c0a1e5326d55218de33e6a7682f26
https://github.com/wpwrak/solvespace/commit/a77cedbfd8dedd147fb59ebf9af936f4259ff31b
https://github.com/wpwrak/solvespace/comm [... cut]
commit 3eaa99cb (2022-01-08): Make all points shown in the Property Browser into active links

This makes all points explorable/selectable and fixes #165 partially.

This is a clean implementation that follows the style of the UI code for
the text window. Previous unmerged attempts from the above issue are:
https://github.com/Evil-Spirit/solvespace-master/commit/eb3db32059bc5b3a4264aba7524ea1cd3e39b8f3
https://github.com/wpwrak/solvespace/commit/782a5bbbe61c0a1e5326d55218de33e6a7682f26
https://github.com/wpwrak/solvespace/commit/a77cedbfd8dedd147fb59ebf9af936f4259ff31b
https://github.com/wpwrak/solvespace/commit/41e55b3d79 [... cut]
closed 2022-01-08 by phkahler