solvespace/solvespace #167: Can't make periodic cubic

mdejean, 2017-01-24:
regression in [55ae10b](https://github.com/solvespace/solvespace/commit/55ae10b5b84ce51042ac9957cfbf01e70d48c952). you're supposed to be able to click again on the first point of a cubic to make a periodic cubic ([HitTestMakeSelection allows this](https://github.com/solvespace/solvespace/blob/55ae10b5b84ce51042ac9957cfbf01e70d48c952/src/draw.cpp#L402)), but [this](https://github.com/solvespace/solvespace/commit/55ae10b5b84ce51042ac9957cfbf01e70d48c952#diff-d1e73a0157de46c51742e677b04e8c0c) prevents selecting the point.

Evil-Spirit, 2017-01-24:
@whitequark, fixed here.

whitequark, 2017-01-24:
Actually, I can't seem to be able to reproduce the bug.
  * Select "Sketch → Bezier Cubic Spline"
  * Place a 1st point
  * Place a 2nd point
  * Place a 3rd point
  * Click the 1st point again
... creates a periodic cubic for me here.

mdejean, 2017-01-24:
Ah, yeah. [this](https://github.com/solvespace/solvespace/commit/55ae10b5b84ce51042ac9957cfbf01e70d48c952#diff-f43ee4b722fdb04241093b055c6cb28dR1044) adds the cubic to pending.requests, but then immediately calls ClearSuper() which empties pending.requests, preventing the bug from appearing.

whitequark, 2017-01-24:
Fix merged in master.

closed 2017-01-24 by whitequark