solvespace/solvespace #171: coincident edges result in NURBS failure

whitequark, 2017-01-25:
Open [bug.zip](https://github.com/solvespace/solvespace/files/728328/bug.zip). This file, extracted as an example, shows an instance of a very pervasive NURBS issue: performing an union or a difference with surfaces that have coincident edges, but do not have Bezier control points in exactly the same place in both edges, result in a failure.

@Evil-Spirit do you have any idea, at least, why this happens?

Evil-Spirit, 2017-01-25:
@whitequark,
The reason can be is SBezier::Equals. When we have found exact intersecton of cylindrical surface and top cap of different bodies, we are adding this as intesection curve (AddExactIntersectionCurve). While we process it, we are trying to find some identical curve, but comparation is performed only for contorl points, not like integrated length of the difference of the radius-vectors: S( (a(t)-b(t)).Magnitude() ) dt.

whitequark, 2017-01-25:
@Evil-Spirit can you fix this?

Evil-Spirit, 2017-01-25:
@whitequark, I can try figure out possibility of the fixing.

whitequark, 2017-01-25:
@Evil-Spirit please do.

Evil-Spirit, 2017-01-25:
@whitequark, OK