solvespace/solvespace #173: select underlying line segement

Leichtbier, 2017-01-28:
### System information

SolveSpace version: 2.3~7c1ca460

Operating system: Win10

I'm not able to select the underlying line segement if two line segemnts are upon each other.
I can not hide the first selection to select the underlying line segement. I'm not able to select both to choose. The only way is to search the list "requests in group". Is there are an other way to select an underlying line segement?

wpwrak, 2017-01-28:
Now that we can select such things from the text window, perhaps a selection of multiple items could could make the items in question also selectable in the text window, with the option of de-selecting ? That way, an ambiguous selection could just select everything applicable, and the user could get rid of the undesired bits.

Another way for handling ambiguous selections that works quite nicely (but would require some changes to the selection logic if applied to SolveSpace) is something I've implemented in fped:
- first, make a list of the candidates, sorted roughly by the difficulty of catching the respective item,
- tentatively select the one that's hardest to catch,
- if clicking again on the same selection, pick the item after the one currently selected,
- and so on.

whitequark, 2017-01-28:
We even should already have the list of entities under cursor.

Evil-Spirit, 2017-01-29:
This even was already implemented as "select by order"

whitequark, 2017-01-29:
I'm not sure if that's the best UI really, compared to e.g. listing all entities in the text window. I'll need to think about it.

Evil-Spirit, 2019-08-21:
https://github.com/Evil-Spirit/solvespace/commit/a7be01991d4d30556d1442d5d3c5da13b67bcbe9

phkahler, 2020-11-21:
@Leichtbier and  @wpwrak  There is a pending pull request to give preference to non-construction lines when selecting entities ( #810 ). What do you think of that? Would that be better than having the current behaviour which seems more random?

phkahler, 2021-05-16:
FYI It is now possible to pick coincident lines by slightly rotating the sketch. For depth comparison each line is considered to be at the depth of its midpoint. That means if you rotate the sketch a little bit it will select the "closest" line when they overlap.

phkahler, 2021-11-11:
Current master also has the exploded view for 2d sketches, which separates all lines by an adjustable interval perpendicular to the sketch plane.

phkahler, 2021-12-07:
I'm going to close this, as the exploded view should cover it. There have been other improvements to selection since this was opened as well, and there has been no feedback on any of that.

closed 2021-12-07 by phkahler