solvespace/solvespace #174: distances can be hard to drag

wpwrak, 2017-01-28:
Lately, I'm having a lot of difficulties dragging distances. Selecting them is fine. I'm not entirely sure if it's because I unlearned some movement that happens to work better, or simply because my designs are getting more crowded.

When trying to drag a distance, I typically aim first for the text. If that doesn't work, I try to get the line(s) to the item being sized.

Perhaps the dragging function could, when deciding what to do with the drag movement, give preference to the item already selected.

whitequark, 2017-01-28:
@wpwrak do you have any sketch where this is reproducible? Please also post a screenshot with position/zoom settings that reproduce the issue.

wpwrak, 2017-01-28:
Here is one:
http://downloads.qi-hardware.com/people/werner/slvs/drag-diam-hard.png
The cursor (alas, xwd can't show it) is over the center of the circle-with-slash. Pressing the left button and dragging does nothing.

The .slvs is simple enough:
http://downloads.qi-hardware.com/people/werner/slvs/drag-diam-hard.slvs

whitequark, 2017-01-28:
confirmed

Evil-Spirit, 2019-08-21:
https://github.com/Evil-Spirit/solvespace-master/commit/3c8cc2e3cf22aad8546da4682938077df08e7361

baryluk, 2020-01-16:
Same here. I have a line-distance constrain selected, but I can't drag it, because it is overlapping with line behind it I guess. It happens even if the constrain is a reference value.

I am using git master 16c5fa6889202f2059b52c4a89599910541dd1a0

baryluk, 2020-01-16:
@Evil-Spirit I can confirm that the patch from https://github.com/Evil-Spirit/solvespace-master/commit/3c8cc2e3cf22aad8546da4682938077df08e7361 fixes the issue.

I do not know if it is the correct solution, but it does fixes this particular problem for me.

The patch is not really in mergable state tho. It should just remove the commented code, not comment it out. :D Or fix the logic there, otherwise it looks a bit more like a hack.

phkahler, 2020-06-24:
IMHO that patch is no good. It eliminates the ability to drag entities from prior groups. Normally if you extrude a rectangle you can push or pull the "new" end to change the length, while dragging points or lines of the original rectangle can change the width or height. The patch eliminates the option for the later - resizing the underlying sketch from the extrusion group.

phkahler, 2020-06-24:
BTW if it's the same this is really easy to reproduce. Draw a circle, press "d" and type in 10. The diameter will be positioned right in the middle of the circle and will be really hard to move. It highlights correctly when hovered, but often will refuse to drag. Is worse with small circles, hence typing 10 for diameter on default view.

phkahler, 2020-08-09:
Can anyone confirm the fix in PR #672  fixes this on Windows and OSX? It's good on Linux.

nabijaczleweli, 2020-08-09:
Confirmed on Windows.

closed 2020-08-09 by phkahler

phkahler, 2020-08-10:
@nabijaczleweli Thanks!