solvespace/solvespace #175: forcing NURBS to mesh causes "naked edges"

wpwrak, 2017-01-28:
SolveSpace at commit d4b052d34d94a62d9a9921f9c7916c4821bb325a

The following design produces complaints about 864 naked edges, which makes SolveSpace very unhappy:
https://gitlab.com/anelok/mexp/raw/016367ea70960daac4d8bb94a5740d84a3f5e900/alt13/aaa-minus-weirdly-broken.slvs

I found that turning off NURBS to mesh in g00b-spring makes the naked edges go away.

Before I noticed that I had NURBS to mesh selected, I tried removing the rectangular blocks on the side, the hole, the "dimple", ... none of which helped to get rid of the naked edges.

whitequark, 2017-02-05:
Triage: still an issue after baf9dc0aae573f017b8477e5d438665e90e2d5ae, although I get many fewer problematic edges (different chord tolerance?).

whitequark, 2017-03-12:
Triage: per discussion in https://github.com/Evil-Spirit/solvespace-master/commit/51616bc339d4a980dc2ae73c8a140be0fdefdcc0#commitcomment-21277921, switching from ear-triangulation to Delaunay will help here.

Evil-Spirit, 2017-09-29:
@whitequark 
What about using https://www.geometrictools.com/Samples/Geometrics.html?
Occasionally, my friend @bazhenovc made cmake building for this https://github.com/bazhenovc/WildMagic/commits/master

Evil-Spirit, 2017-09-29:
@bazhenovc
What do you know about this library? Is this the best choice?

bazhenovc, 2017-09-29:
Oh hello there.

I can say that this library does the job and the code is of high quality - so yes, I'd recommend it. I'd suggest to extract the needed classes rather then using the whole WM5/GTE - it's quite huge and you'll probably never need the whole package.

whitequark, 2017-09-29:
@bazhenovc Thanks for chiming in!

@Evil-Spirit This uses the Boost license, so it's compatible. It will remove a massive chunk of code from SolveSpace, if I understand correctly (exactly which?). Seems easy to integrate, though I think using a submodule is preferable to copying files (history is lost, upgrades are painful, etc).

bazhenovc, 2017-09-29:
History will be lost no matter what you'll do - their source code is distributed in zip archives.

whitequark, 2017-09-29:
In these cases I normally import releases into git version by version (which is quite easy with a script), so we can still easily bisect, and also easily track/rebase our changes on top of their classes.

Evil-Spirit, 2017-09-29:
@bazhenovc 
Thanks for information!

@whitequark 
At least we can use Delaunay triangulation.

whitequark, 2017-09-29:
Anything else?

Evil-Spirit, 2017-09-30:
there is CORK, some library for booleans.
Not sure this is production ready
http://arc-team-open-research.blogspot.ru/2014/10/boolean-operations-powerful-cork.html

whitequark, 2017-09-30:
There's this issue with it:

> WARNING: Unfortunately, there are a number of known problems with Cork. I have zero time to work on the library since I'm currently busy working on a thesis. (March 2016) If you would like to pay me large sums of money to improve this library, maybe we should talk. Otherwise, I'm really quite sorry to say I don't have time.

phkahler, 2020-08-13:
The changes in #673 seem to fix the NURBS booleans here, so using triangle mesh is not necessary after that fix.

phkahler, 2020-09-09:
Using current master (dbf5a8d) with default setting this gives me 8 bad edges. Switching it back to NURBS produces zero bad edges even if I increase max_segments to 100 and lower chord tolerance to 0.01 percent. It is always best **not** to use triangle mesh unless something goes wrong with a NURBS boolean operation. Maybe at the time this issue was opened that was necessary.

This is still an issue and a fairly simple test sketch.

phkahler, 2022-02-06:
I just revisited this model with commit 61cc28f8b2  (late in 3.0 prior to 3.1) and the problem is still there. I noticed that switching to NURBS and then back to triangle mesh often changes the number of bad edges. Anywhere from zero up to 126 but typically about 30. There seem to be some randomness in the boolean operations on triangle meshes.