solvespace/solvespace #176: placing points on lines in 3D makes them appear at strange places

wpwrak, 2017-01-31:
When placing a point on a line in 3D, it can appear at a different location than the one where I clicked. Here is an example showing a moderate displacement:
[image]

The model is
https://gitlab.com/anelok/mexp/raw/88480766568dc2ab7c00d2d0bea628c9895e554e/alt13/an-top.slvs

whitequark, 2017-01-31:
That's expected behavior, really, in any view that's not parallel to the line you're placing the point at.

wpwrak, 2017-01-31:
But why not place it where I asked, instead of making me search for where it may have ended up ? Especially if there are a lot of points (like in the example above), that can get quite tricky. Given that SolveSpace recognized the automatic point-on-line constraint, there is a valid position for that point in the immediate vicinity of the click location.

Evil-Spirit, 2017-01-31:
@whitequark, We need to implement ModifyToSatisfy for pt-on-line and call it when we are creating like for distances (since pt-on-line has hidden parameter which needs to be guessed when we just create constraint)

whitequark, 2017-01-31:
@Evil-Spirit ok

Evil-Spirit, 2017-02-02:
@whitequark, update here

whitequark, 2017-02-05:
Depends on #172

Evil-Spirit, 2017-02-06:
@whitequark, also here

Evil-Spirit, 2017-02-06:
@whitequark, sorry, not working here. wait.

Evil-Spirit, 2017-02-06:
@whitequark, ready here

whitequark, 2017-02-06:
Fixed in master.

closed 2017-02-06 by whitequark