solvespace/solvespace #177: naked edges vs. chord tolerance

wpwrak, 2017-01-31:
Given this design,
https://gitlab.com/anelok/mexp/raw/8951232e87fa6c7a8f2820436a2ae9c247dd26a6/alt13/an-top.slvs
all is well with chord tolerance 0.1.

However, if I reduce it to 0.05, suddenly naked edges appear:
[image]

The history of plus-base is as follows: I first made sk-minus-base and extruded it to minus-base. Then I added the workplane for sk-plus-base, paste the content of sk-minus-base into it, reattach the now loose constraints, then extrude. Not sure if the copy & paste plays any part in the appearance of naked edges, but this is one thing I didn't do anywhere else.

I also deleted and reconstructed sk-plus-base, this time based on g006-an-side-overlap, but that produced exactly the same result.

Evil-Spirit, 2017-02-02:
@wpwrak, I have cloned your repo and switched out this revision. But I can't reproduce this. I have no problem for export chord tolerances: 0.1, 0.05, 0.01. Can you switch and check again? Also, what SolveSpace  revision do you have?

wpwrak, 2017-02-02:
I'm running a mix of master plus some cherry-picked changes, but I can also make it happen with solvespace/solvespace master (279424)
Build options:
cmake .. -DENABLE_TESTS=OFF -DCMAKE_BUILD_TYPE=Release
(but it also happens with just cmake .. -DENABLE_TESTS=OFF)

Evil-Spirit, 2017-02-04:
@wpwrak, I still can't reproduce. Can you attach exported STL file?

wpwrak, 2017-02-04:
Alright, here is the setup:
-  solvespace/solvespace master at 4465bc0270ee185e5029ecacfe27665fa7fe5ab0
-  cmake .. -DCMAKE_BUILD_TYPE=Release
-  anelok/mexp at 8951232

SolveSpace configuration:
-  chord tolerance 0.10% / 0.10 mm
-  max piecewise linear segments 100
-  export chord tolerance 0.10
-  export max piecewise 64

First test:
-  Analyze > Show Naked Edges: all is well
-  Export Triangle Mesh: all is well
-  http://downloads.qi-hardware.com/people/werner/slvs/177-10.stl

Change configuration:
-  chord tolerance 0.05% / 0.05 mm
-  export chord tolerance 0.05

Second test:
-  Analyze > Show Naked Edges: 17 problematic edges;
[image]
-  Export Triangle Mesh: same complaint
  http://downloads.qi-hardware.com/people/werner/slvs/177-05.stl

wpwrak, 2017-02-04:
@Evil-Spirit: by the way, with the version I use for actual work, which is a mix of master, DISPLAY_TEXTWND, your two performance enhancements, the #186 fix, and explorability enhancements, I get only 11 naked edges:
[image]

The resulting STL:
http://downloads.qi-hardware.com/people/werner/slvs/177-05-mishmash.stl

Evil-Spirit, 2017-02-04:
@wpwrak, thank you, I will try to figure out what happens.

Evil-Spirit, 2017-02-05:
@wpwrak, can you try this? https://github.com/Evil-Spirit/solvespace-master/commit/51616bc339d4a980dc2ae73c8a140be0fdefdcc0

Evil-Spirit, 2017-05-22:
@whitequark 5 problematic edges here

whitequark, 2017-05-22:
is that after the fix in #259?

Evil-Spirit, 2017-05-22:
@whitequark hm. probably no progress here. it is like was before.

closed 2023-08-09 by phkahler