solvespace/solvespace #178: Project 2D sketch onto 3D surface

kellpossible, 2017-01-31:
I've an application where it is necessary to project a 2d sketch onto a 3d surface

whitequark, 2017-01-31:
Even if the projection operation itself would be implemented (and that's already a problem for the simple reason that we have a mesh backend alone), the resulting curves won't be analytic, and won't work with our constraint system, so that doesn't work.

closed 2017-01-31 by whitequark

kellpossible, 2017-02-01:
oh, nevermind I didn't realise this would be impossible due to fundamental limitations of the 3d implementation, thanks anyway. Surely some of the other CAD packages support analytic projected curves?

whitequark, 2017-02-01:
> Do some of the other CAD packages support analytic projected curves?

None come to mind, but I haven't worked a lot with the large commercial packages. In general this seems like a really hard problem to solve.

Evil-Spirit, 2017-02-02:
@kellpossible, Any CAD can't handle analytical NURBS intesection. NURBS is an illusion of exact representation, math guys can't solve NURBS surface intersection in analytically-exact way. So we have just use numeric approximation.

kellpossible, 2017-02-02:
@Evil-Spirit and yet we would say that NURBS itself is analytic? And the curves themselves are analytic. Perhaps there would be a way to define the intersection path as a function of the two, and evaluate it numerically to get an adaptive analytic approximation of the result?

kellpossible, 2017-02-02:
@whitequark it seems so, I'll have a google around to see if there's any good papers on the subject. perhaps room for new research? wish I was better at maths

Evil-Spirit, 2017-02-02:
@kellpossible, It is common to use rational cubic bezier for both surface and cure representation, and yes, for this case all is analytical and exact. But when we are projecting cubic curve onto cubic surface, we can't get cubic curves anymore. This is the first. The second... We can't even found this intersection in analytical form. But if you doesn't mean cubic on cubic but just project anything onto plane - this works perfect now.

kellpossible, 2017-02-02:
@Evil-Spirit 
I was thinking something more like this, where you project a curve onto another curved object with a plane of projection:
[image]
[image]

projected_curve = function( .... oh I see there isn't an analytic function for the surface position in terms of projection line intersection.

Perhaps something like this instead?

projected_curve = numeric_solve_projected_position_on_surface(curve_to_be_projected(t), plane_of_projection)

It could be calculated accurately and then compressed using bezier curves at a lower resolution perhaps?

Evil-Spirit, 2017-02-03:
@kellpossible, Yes, this is how this problem solved in other CADs - compute approximation numerically and try to interpolate it using bezier curves.

mirecta, 2018-04-22:
so no news here ? if i want put some text or something extrude on curved surface how i can  do it ?

whitequark, 2018-04-22:
You can't.