solvespace/solvespace #179: "Unselect hovered" is broken

whitequark, 2017-02-01:
It just never appears anymore.

Evil-Spirit, 2017-02-01:
@whitequark, this works for me

whitequark, 2017-02-01:
@Evil-Spirit Try this:
  1. New file
  2. Draw a rect
  3. Click on top line
  4. Hover over top line and right-click

whitequark, 2017-02-01:
@Evil-Spirit Oh... I've just realized "Unselect Hovered" doesn't appear unless more than one entity is selected. Nevermind.

cc @wpwrak

wpwrak, 2017-02-01:
Hmm, but even if selecting two lines of the rect, I don't see any "Unselect Hovered", just "Unselect All":
[image]

whitequark, 2017-02-01:
@wpwrak Are you really hovering the top line? The menu seems quite far from it.

wpwrak, 2017-02-02:
Yes, the line pales with fear as the mouse pointer approaches. Alas, it turns regular red again when right-clicking, so one can't see that in the screenshot.

By the way, for this type of debugging, it may be useful to have an option somewhere (could be on the command line), to make SolveSpace draw a cursor (maybe X-shaped), so that it is included in the screenshot.

whitequark, 2017-02-02:
> By the way, for this type of debugging, it may be useful to have an option somewhere (could be on the command line), to make SolveSpace draw a cursor (maybe X-shaped), so that it is included in the screenshot.

This is really the responsibility of your screenshot program. Both GNOME and KDE provide such an option, for one.

wpwrak, 2017-02-02:
Sigh, all that modernism (using xwd here) ... here we have one with cursor, courtesy of "screengrab":
[image]

whitequark, 2017-02-17:
@wpwrak can you please use gdb to trace GraphicsWidget::{on_motion_notify_event,on_button_press_event,on_button_release_event} calls?

whitequark, 2019-05-23:
@wpwrak can you still reproduce this?

whitequark, 2019-07-31:
No response in two years, closing.

closed 2019-07-31 by whitequark