solvespace/solvespace #181: Linking latest master fails on macOS

virtualritz, 2017-02-02:
```
[ 97%] Linking CXX executable ../bin/solvespace.app/Contents/MacOS/solvespace
Undefined symbols for architecture x86_64:
  "SolveSpace::SolveSpaceUI::ExportMeshTo(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)", referenced from:
      SolveSpace::SolveSpaceUI::MenuFile(SolveSpace::Command) in solvespace.cpp.o
  "SolveSpace::SolveSpaceUI::ExportAsPngTo(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)", referenced from:
      SolveSpace::SolveSpaceUI::MenuFile(SolveSpace::Command) in solvespace.cpp.o
  "SolveSpace::SolveSpaceUI::ExportSectionTo(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)", referenced from:
      SolveSpace::SolveSpaceUI::MenuFile(SolveSpace::Command) in solvespace.cpp.o
  "SolveSpace::SolveSpaceUI::ExportViewOrWireframeTo(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, bool)", referenced from:
      SolveSpace::SolveSpaceUI::MenuFile(SolveSpace::Command) in solvespace.cpp.o
  "SolveSpace::VectorFileWriter::BezierAsPwl(SolveSpace::SBezier*)", referenced from:
      SolveSpace::HpglFileWriter::Bezier(SolveSpace::SBezier*) in libsolvespace-core.a(exportvector.cpp.o)
      SolveSpace::GCodeFileWriter::Bezier(SolveSpace::SBezier*) in libsolvespace-core.a(exportvector.cpp.o)
  "SolveSpace::VectorFileWriter::BezierAsNonrationalCubic(SolveSpace::SBezier*, int)", referenced from:
      SolveSpace::EpsFileWriter::Bezier(SolveSpace::SBezier*) in libsolvespace-core.a(exportvector.cpp.o)
      SolveSpace::PdfFileWriter::Bezier(SolveSpace::SBezier*) in libsolvespace-core.a(exportvector.cpp.o)
      SolveSpace::SvgFileWriter::Bezier(SolveSpace::SBezier*) in libsolvespace-core.a(exportvector.cpp.o)
  "SolveSpace::VectorFileWriter::MmToPts(double)", referenced from:
      SolveSpace::EpsFileWriter::StartFile() in libsolvespace-core.a(exportvector.cpp.o)
      SolveSpace::EpsFileWriter::FinishPath(SolveSpace::RgbaColor, double, bool, SolveSpace::RgbaColor, SolveSpace::hStyle) in libsolvespace-core.a(exportvector.cpp.o)
      SolveSpace::EpsFileWriter::MaybeMoveTo(SolveSpace::Vector, SolveSpace::Vector) in libsolvespace-core.a(exportvector.cpp.o)
      SolveSpace::EpsFileWriter::Triangle(SolveSpace::STriangle*) in libsolvespace-core.a(exportvector.cpp.o)
      SolveSpace::EpsFileWriter::Bezier(SolveSpace::SBezier*) in libsolvespace-core.a(exportvector.cpp.o)
      SolveSpace::PdfFileWriter::StartFile() in libsolvespace-core.a(exportvector.cpp.o)
      SolveSpace::PdfFileWriter::StartPath(SolveSpace::RgbaColor, double, bool, SolveSpace::RgbaColor, SolveSpace::hStyle) in libsolvespace-core.a(exportvector.cpp.o)
      ...
  "SolveSpace::VectorFileWriter::Transform(SolveSpace::Vector&) const", referenced from:
      DxfWriteInterface::writeEntities() in libsolvespace-core.a(exportvector.cpp.o)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)

```

whitequark, 2017-02-02:
please `make clean` then build again

whitequark, 2017-02-12:
@virtualritz ping?

closed 2017-03-04 by whitequark