solvespace/solvespace #182: undo deletion leads to alarming complaints

wpwrak, 2017-02-02:
I have a design where trying to undo a deletion causes an alarming complaint about redundant constraints. It <I>seems</I> that there is no real problem in the end, but it's still unsettling.

Unfortunately, this is a complex scenario. It is based on
https://gitlab.com/anelok/mexp.git
as of commit 7c1ff0aa55dbacc7cdede37aa8b24a1faf9a74a5 (master).

To reproduce,
1) open alt13/an-top.slvs
2) activate g00a-sk-panel-cut
3) select r010-line-segment (the vertical line on the left)
4) delete it. We get a lot of damage: 8 requests, 26 constraints, 4 groups.
5) Edit > Undo
6) text window reports a redundant constraint in g02d-balance
7) "home" shows a number of groups (g00b, g020, g029, g02b, g02d) with red "NO"
8) activating any group clears the red "NO"s up to that group

Especially 6) is unexpected and alarming, because, instead of returning to safety, it suggests that uncontrolled damage has occurred.

whitequark, 2017-02-06:
@Evil-Spirit Looks like replacing 

    SS.GenerateAll(SolveSpaceUI::Generate::ALL);

with

    SS.GenerateAll(SolveSpaceUI::Generate::REGEN);

fixes NG groups *after* the active one but breaks a group *before* the active one instead.

Evil-Spirit, 2017-02-06:
@whitequark, hm.. strange behaviour.

Evil-Spirit, 2017-02-15:
@whitequark, I have finally figure out what's happens here. Look at https://github.com/solvespace/solvespace/blob/master/src/group.cpp#L783 - the faces for top/bottom extrude is created here. For this faces we should take normal somewhere and this normal is taken from assembled loop (which is not exist for the first pass of generatation since it is generated here https://github.com/solvespace/solvespace/blob/master/src/generate.cpp#L279). But generally this is not all of problems because loops can be non-existant since there are no loops after deletion of one of the lines. Probably, we should save us from the both cases, but this can't be performed for sketches in 3d.

here is isolated case: https://github.com/whitequark/solvespace-testcases/blob/master/issues/182/182-isolated.slvs

Evil-Spirit, 2017-02-15:
@whitequark, here is the fixes, but it breaks some tests, I am trying to fix them. 
@jwesthues, What is your opinion on https://github.com/Evil-Spirit/solvespace-master/commit/c677df188c9863c9828c9a465cbc69192f25d95c?

jwesthues, 2017-02-15:
I guess it would be slightly better to use the normal from the loops if one exists (even in workplane groups), otherwise the workplane's normal? The cases where that matters are pretty pathological, though.

Evil-Spirit, 2017-02-15:
@jwesthues, what can you say about equality of loops and workplane normal? Is it always equal, or this can look at different directions? My thoughts was about introducing stability in normal direction (not let it swap direction), so what arguments for keep using loop normal? Can using normal from workplane break any leagcy files?

Evil-Spirit, 2017-02-15:
@whitequark, reimplemented in safe way as @jwesthues suggested.

jwesthues, 2017-02-15:
Nothing specifically requires you to sketch in the workplane group's own workplane. That's a little pathological, but your change addresses that.

whitequark, 2017-02-17:
Fixed in master.

closed 2017-02-17 by whitequark