solvespace/solvespace #184: problem with text input field on Arch Linux

GerhardPaulus, 2017-02-02:
solvespace on Arch Linux (both current) :

text input field in GraphicsWindow and TextWindow gets black transparent background, as soon as you start to type. Editing is effectively black on black.

The simplest workaround I found is commenting out the line in src/platform/gtkmain.cpp where the frame of Gtk::Entry is removed. 

class EditorOverlay : public Gtk::Fixed {
public:
    EditorOverlay(Gtk::Widget &underlay) : _underlay(underlay) {
        set_size_request(0, 0);
        add(_underlay);
        _entry.set_no_show_all(true);   
//        _entry.set_has_frame(false);
        add(_entry);
        _entry.signal_activate().
            connect(sigc::mem_fun(this, &EditorOverlay::on_activate));
    }

I checked with _entry.get_style_context()->get_background_color().to_string()  before and after _entry.set_has_frame(false) and this instruction currently changes the standard background color of this widget from  opaque white rgb(255,255,255) to transparent black rgba(0,0,0,0) . 

Gerhard

whitequark, 2017-02-05:
@GerhardPaulus Unfortunately `set_has_frame(true)` screws up positioning.

The correct patch should be somewhere along these lines:
```
diff --git a/src/platform/gtkmain.cpp b/src/platform/gtkmain.cpp
index 079fd395..5bab2dec 100644
--- a/src/platform/gtkmain.cpp
+++ b/src/platform/gtkmain.cpp
@@ -17,6 +17,7 @@
 #include <glibmm/convert.h>
 #include <giomm/file.h>
 #include <gdkmm/cursor.h>
+#include <gtkmm/cssprovider.h>
 #include <gtkmm/drawingarea.h>
 #include <gtkmm/glarea.h>
 #include <gtkmm/scrollbar.h>
@@ -1417,6 +1418,17 @@ int main(int argc, char** argv) {
 
     Gtk::Main main(argc, argv);
 
+    // Add our application-specific styles, to override GTK defaults.
+    Glib::RefPtr<Gtk::CssProvider> style_provider = Gtk::CssProvider::create();
+    style_provider->load_from_data(R"(
+    GtkEntry {
+        background: red;
+    }
+    )");
+    Gtk::StyleContext::add_provider_for_screen(Gdk::Screen::get_default(),
+                                               style_provider,
+                                               600 /*Gtk::STYLE_PROVIDER_PRIORITY_APPLICATION*/);
+
 #ifdef HAVE_SPACEWARE
     gdk_window_add_filter(NULL, GdkSpnavFilter, NULL);
 #endif
```
... but unfortunately, because GTK 3 is a pile of crap, this doesn't do anything and I couldn't permute it to make it work. If you feel adventurous you can fix this, otherwise I am not going to waste any more time fighting GTK.

GerhardPaulus, 2017-02-15:
I installed a GTK theme and text input is now black on white.

closed 2017-02-15 by GerhardPaulus