solvespace/solvespace #187: Exported DXF files are unreadable by QCad, A360, DraftSight

whitequark, 2017-02-03:
See this thread: http://solvespace.com/forum.pl?action=viewthread&parent=1741.
I suspect 99d6f1f is the cause but don't understand why.

Evil-Spirit, 2017-02-04:
@whitequark, may be this can help... https://github.com/Evil-Spirit/libdxfrw/commit/f3f38c723d1e333e871b5f1a7360a26aa41f81ec

Evil-Spirit, 2017-02-04:
@whitequark, with inkscape still the problem.. I can't find where dxf import source of inkscape. This can help.

whitequark, 2017-02-04:
@Evil-Spirit It's a bunch of disgusting Python code here: https://github.com/inkscape/inkscape/blob/bzr-original/share/extensions/dxf_input.py

whitequark, 2017-02-05:
@Evil-Spirit I merged your fix, except I don't mark the POLYLINE as 3d if every point has Z exactly at zero. This still doesn't work with Inkscape, probably because its DXF import is badly written. We have SVG export anyway, so I don't consider this a priority.

whitequark, 2017-02-05:
This fixed DraftSight and QCad, so I believe that's the root cause. Let's backport this to 2.x...

whitequark, 2017-02-06:
Apparently still some problems in QCad... http://solvespace.com/forum.pl?action=viewthread&parent=1741

Evil-Spirit, 2017-02-06:
@whitequark, I can't reproduce it for QCad. Probably, not the same version or somthing else. I have BircsCAD, NanoCAD, DWG TrueView, LibreCAD, QCad, Inkspace installed and perform testing with them...

whitequark, 2017-02-06:
@Evil-Spirit Can you confirm that export works properly with everything except Inkscape??

Evil-Spirit, 2017-02-06:
@whitequark, OK I am testing with the last master now... Wait 10 minutes.

Evil-Spirit, 2017-02-06:
@whitequark, All is working except Inkscape.
`1 ENTITIES of type POLYLINE encountered and ignored. Please try to convert to Release 13 format using QCad.`

whitequark, 2017-02-06:
Which version of QCad are you using?

Evil-Spirit, 2017-02-06:
@whitequark, 3.13 free version

Evil-Spirit, 2017-02-06:
@whitequark, may be I should try install commercial trial...

whitequark, 2017-02-06:
@Evil-Spirit in Qcad 3.16.4 (trial) actually I don't get a rectangle at all.

Evil-Spirit, 2017-02-06:
@whitequark, I have no this problem in windows. Can you attach dxf?

Evil-Spirit, 2017-02-06:
[image]

whitequark, 2017-02-06:
@Evil-Spirit Oh, interesting. I found that Qcad misreads SolveSpace's DXF files, but *only* when the paid addon is activated.

As for a file, just draw a single rectangle, that was enough to me.

Evil-Spirit, 2017-02-06:
@whitequark, So you are meaning what this can be just QCad bug?

whitequark, 2017-02-06:
Maybe... Teigha is produced by ODA, which includes Autodesk. Can you try https://www.opendesign.com/guestfiles/TeighaFileConverter ?

Evil-Spirit, 2017-02-06:
Teigha is used by BricsCAD and nanoCAD. They are open files smoothly...

whitequark, 2017-02-06:
OK, then it's a QCad bug.

closed 2017-02-06 by whitequark