solvespace/solvespace #188: Exported DXF files are not readable for RDWorks

Leichtbier, 2017-02-04:
Since SolveSpace 2.1 I'm no longer able to use the exported DXF files with RDWorks (a software for my laser cutter).

You can get the software here:
[http://www.thunderlaser.com/download/down/RDWorksV8.rar](url)

I had the same problem with inkscape, but that seems to be fixed. My current workaround is to save as svg and than convert to dxf.
Solvespace is a realy nice programm fore laser cutting and a even nicer with this functionality ^^

Evil-Spirit, 2017-02-04:
@Leichtbier, I have tried this and have no problem to import dxf in RDWorks.
But with Inkspace still the trouble... You can also look here https://github.com/solvespace/solvespace/issues/187

Evil-Spirit, 2017-02-04:
@Leichtbier, can you attach slvs example and way to reproduce bug?

Evil-Spirit, 2017-02-04:
[image]

whitequark, 2017-02-12:
@Leichtbier ping?

Leichtbier, 2017-02-12:
Pong ^^ I've tryed it yesterday with the newest version of RDWorks and now it works again. Thank you for soliving my problem.

closed 2017-02-12 by whitequark