solvespace/solvespace #189: 1:1 on screen

Leichtbier, 2017-02-05:
In the 3d view parameters is the parameter 'overall scale factor'. It could be handy to have also an additional default parameter for this parameter combined with a shortcut key to set the overall scale factor to this default value. Than you can adjust your default parameter so the scale on your screen is 1:1. Furthermore the unit for 'overall scale factor' could than be percent.

whitequark, 2017-02-05:
You can already get 1:1 scale on screen by pressing the "set to full scale" link on the view properties. I don't think this is an operation common enough to warrant a shortcut, seeing as we're short on those already.

closed 2017-02-05 by whitequark

Leichtbier, 2017-02-05:
In Inkscape you press '1' and you get a scale of 100%. In solvespace you have no echo of your current scale factor until you are in the '3d view parameters'.

whitequark, 2017-02-05:
Sure. I just don't see why is this such a common need as to warrant a shortcut.