solvespace/solvespace #191: undefined symbol: g_simple_io_stream_get_type

holocronweaver, 2017-02-06:
### System information

SolveSpace version: master branch as of 2017-02-06 07:47:45 UTC.

Operating system: Ubuntu 16.04 x64

### Expected behavior

What should have happened?

On a fresh build,

`cd solvespace/build;
./bin/solvespace`

Program should run.

### Actual behavior

What actually happened?

An error message:

`GLib-GIO-Message: Using the 'memory' GSettings backend.  Your settings will not be saved or shared with other applications.
./solvespace: symbol lookup error: /usr/lib/x86_64-linux-gnu/libgiomm-2.4.so.1: undefined symbol: g_simple_io_stream_get_type`

whitequark, 2017-02-06:
This is *definitely* something up with your system. Is it fully upgraded? Are there no APT conflicts?

I've just did a fresh build with 16.04 to confirm and it indeed works.

holocronweaver, 2017-02-06:
Reinstalled the library and now solvespace runs fine. No APT conflicts detected though, not sure how it got overwritten.

closed 2017-02-06 by holocronweaver

whitequark, 2017-02-06:
I believe Anaconda specifically avoids altering LD_LIBRARY_PATH because of this problem. Not sure what happened.

holocronweaver, 2017-02-07:
This was bugging me because Anaconda has done this to me before. Found the issue in the solvespace binary rpath:
```
objdump -p ./build/bin/solvespace | egrep 'RPATH|RUNPATH';
  RPATH                /home/user/local/apps/anaconda3/lib:
```
What remains a mystery is why CMake is picking up the path. I am not sure how CMake even knows of the path's existence besides $PATH containing `anaconda/bin`.