solvespace/solvespace #192: custom style on comment causes occlusion

wpwrak, 2017-02-08:
solvespace master (1871efa), plus local changes (non-interfering)

Comments are normally not occluded by solids. However, if setting a custom style, they are - and you can't even turn it off.

Reproduce:
- create rectangular solid: File > New, R, extrude
- add comment somewhere in space
- rotate things and observe that the comment is always visible, also when behind the solid
- select comment and assign new style

Occlusion is now applied to the comment. This is independent from the occlusion setting (rightmost icon in text window).

[image]

Expected behaviour: either always show comments (like measurements or when using the default style), or apply the occlusion setting.

whitequark, 2017-02-08:
This is a regression in 3.0; 2.x used to draw comments, even with custom style, over all other geometry.

Evil-Spirit, 2017-02-15:
@whitequark, fixed here

whitequark, 2017-02-17:
Fixed in master.

closed 2017-02-17 by whitequark